nasa/cFE #499: Documentation on CFE_SB_GetMsgTime()/SetMsgTime() needs updated. They do not set/get timestamps for commands (CFE_SB_CMD_HDR_SIZE).

pcooksey, 2020-01-28:
Documentation in `cFS Application Developers Guide.doc` pg. 57:
> Before sending an SB Message to the SB, the Application can update the SB Message Header.  The most common update is to put the current time in the SB Message.  This is accomplished with one of two SB API functions.  The most commonly used function would be CFE_SB_TimeStampMsg().  This API would insert the current time, in the mission defined format with the mission defined epoch, into the SB Message Header.  The other SB API that can modify the SB Message Header time is CFE_SB_SetMsgTime().  This API call sets the time in the SB Message Header to the time specified during the call.  This is useful when the Application wishes to time tag a series of SB Messages with the same time.
>
> Other fields of the SB Message Header can be modified by an Application prior to sending the SB Message.  These fields, and the associated APIs, are listed in the following table:

I didn't see anywhere that it said if the message is a Command Msg, i.e.,  `CFE_SB_CMD_HDR_SIZE` then it will not have a timestamp and therefore these functions return 0.

Code: 
https://github.com/nasa/cFE/blob/f12563897df87630c8043740dd5ede4b83725c90/fsw/cfe-core/src/sb/cfe_sb_util.c#L277-L305

Just went down a rabbit hole trying to figure out why the message header sizes didn't match in my test cases.

skliper, 2020-01-28:
Please utilize the issue templates, more likely to get action on a ticket if all the information is provided.

pcooksey, 2020-01-28:
Updated to follow the feature template.

PR #545 (2020-03-18): Updated cFE App Developer's Guide
**Describe the contribution**
Updated the cFE app developer's guide markdown with changes to resolve issues 499, 496, and 478.  Made a variety of other small content updates and reordered some sections.

Fixes #499, fixes #496, fixes #478 

**Testing performed**
None - documentation only

**Expected behavior changes**
None - documentation only

**System(s) tested on**
None - documentation only

**Additional context**
N/A

**Third party code**
N/A

**Contributor Info - All information REQUIRED for consideration of pull request**
Elizabeth Timmons - NASA/GSFC
commit 71df47f0 (2020-03-23): Fix #499, Clarify Get/SetMsgTime effect on cmd msg

Partial fix (see also #545), comments/documentation only
closed 2020-03-18 by astrogeco