nasa/cFE #523: SB subscription reporting request messages out of family

skliper, 2020-02-20:
**Is your feature request related to a problem? Please describe.**
CFE_SB_EnableSubReportingCmd, CFE_SB_DisableSubReportingCmd, CFE_SB_SendPrevSubsCmd are processed like commands but don't increment the command counter.

Typical pattern is for non-ground, inter-app messages to have separate message IDs from ground commands.

**Describe the solution you'd like**
Make consistent with standard pattern

**Describe alternatives you've considered**
None

**Additional context**
See Hk message processing, or the message processing in Time services that don't increment command counter.

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-04-08:
@CDKnightNASA @tngo67 Does SBN or SBNg use these?  Since they are from another app, preference would be to not use the ground command MID.  Can we coordinate a change?

jwilmot, 2020-04-21:
As I recall these were for SBN to use when interfacing with SB. They are not ground commands. SBN is designed to be (as best we could) a plug-in just like any other apps. We wanted to avoid call backs or other compile/link time hooks.  That said, the internal MID are in the same name space as "regular" commands so need to be allocated and managed but not in the ground system database. They should not increment the command counter +/- as those are for operators, but should send an event if an error occurs.

skliper, 2020-04-21:
I think you are saying they should be a separate MID from the ground commands?  Want them to use the same MID (CFE_SB_SUBSCRIPTION_RPT_CONTROL_MID or whatever), or each use their own?

CDKnightNASA, 2020-04-21:
> As I recall these were for SBN to use when interfacing with SB. They are not ground commands. SBN is designed to be (as best we could) a plug-in just like any other apps. We wanted to avoid call [...]

Nope, SBN listens for events that are generated by subscription activity:

```
    Status = CFE_SB_SubscribeLocal(CFE_SB_ALLSUBS_TLM_MID, SBN.SubPipe,
...
    Status = CFE_SB_SubscribeLocal(CFE_SB_ONESUB_TLM_MID, SBN.SubPipe,
```

skliper, 2020-04-21:
So who/what sends the referenced commands?

CDKnightNASA, 2020-04-21:
> So who/what sends the referenced commands?

Sorry, looking through the code, SBN does send the relevant commands. I was looking for direct calls to the API.

jwilmot, 2020-04-22:
The required SBN behavior is to requests subscriptions when it is done initializing the subnetwork and discovered peers. This can happen any time after local system init or later due to fault management restarting SBN. This decouples SBN from the local system. The internal commands  CFE_SB_EnableSubReportingCmd, CFE_SB_DisableSubReportingCmd, CFE_SB_SendPrevSubsCmd support this. Like other internal commands, the counters should not increment but error events should be sent. Sorry I am being dense, but it's still not clear what problem this ticket is trying to solve.

skliper, 2020-04-22:
Discussed w/ @jwilmot and got concurrence with the following approach.

Inter-app messages/MIDs:
CFE_SB_SEND_HK_MID - For scheduler to request HK message
CFE_SB_SUBSCRIPTION_RPT_CONTROL_MID - Use for these three commands

Ground command MID:
CFE_SB_CMD_MID - ground commands only

So it'll require a 1 line change in SBN, but provides separation between the inter-app and ground commands (preferred model).

closed 2020-05-13 by astrogeco

juliejohn015, 2021-06-16:
[Subscription Reporting](https://www.subscriptionflow.com/subscription-analytics-reporting/) and Analytics Software

PR #647 (2020-05-08): Fix #523, SB Subscription report control on separate MID
**Describe the contribution**
Moves inter-app commands to separate MID (like HK request from SCH).  These requests only come from SBN as of now.  No longer using the ground command MID.  Also updates the unit tests to match the changes (checks new error, updates MID's for command tests).
Fix #523 

**Testing performed**
Build and ran tests local, also CI - https://travis-ci.com/github/skliper/cFS/builds/162454974

**Expected behavior changes**
SBN will need to init command with new MID

**System(s) tested on**
 - Hardware: cFS Dev Server [... cut]
commit 159ab752 (2020-05-08): Merge pull request #647 from skliper/fix523-sb-sub-rpt-mid

Fix #523, SB Subscription report control on separate MID
commit f211a064 (2020-05-08): Fix #523, SB Subscription report control on separate MID
PR #692 (2020-05-13): Integration Candidate COMBINED 2020-04-29 and 2020-05-06
**Describe the contribution**

Fix #399, Deprecate decompress
Fix #411, rework exception handling in CFE
Fix #484, Deprecate output to shell commandFix #491, Add UT macros
Fix #519, Deprecate CFS/FS time conversion APIs
Fix #523, SB Subscription report control on separate MID
Fix #551, Remove duplicate CFE_TIME_Local1HzISR prototype
Fix #559, Resolve doxygen warnings
Fix #618, stubs must not depend on real msgid implementation
Fix #637, Remove old CFE_ARINC653-related tests
Fix #639, Remove ES_APP_DEBUG functionality
Fix #641, string ope [... cut]