nasa/cFE #531: CFE_TBL_NotifyByMessage should send message, not command

skliper, 2020-02-26:
**Is your feature request related to a problem? Please describe.**
CFE_TBL_NotifyByMessage prototype includes a command code.  This implies it's sending a command.  Command processing is typically associated with a requirement to increment the command counter which is typically undesirable for "internally" generated messages.

**Describe the solution you'd like**
Update to send a message, not a command (like hk, or any of the other non-ground commands)

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-02-26:
It actually _is_ sending a command though.  This function itself isn't a command, but it it instructs tbl services to send a command whenever the table gets updated.  

I'm not seeing the issue here....

Note that even "HK" type messages actually do have a command header anyway.  CCSDS doesn't support "bare" packets with no content, we __must__ include the command header, so it must be filled with something.

skliper, 2020-02-26:
This message should not increment the command counter (or invalid if there's a processing issue), and like HK would be better to not use the typical ground command MID.  That's the real restriction.  Probably just a documentation thing.

My concern is probably an artifact of insufficiently defined requirements, where "command" has specific associated requirements when really we need to somehow differentiate between messages that should and shouldn't increment the command counter.  Agreed it still needs to be a "command" type CCSDS message (since TLM definitely doesn't make sense and we only get two choices), but it shouldn't be treated like a ground command.

jphickey, 2020-02-26:
Yes, that's fine but it strikes me more as a system engineering issue not an API issue.  While it's true this notification _could_ send a fully-fledged "command' that increments a command code, that doesn't mean it _must_ do this.

If it really is a concern, just document that the message sent should _not_ be one that increments the command code.  But I don't see any reason to change the API.

skliper, 2020-03-02:
Agreed, as I said...

> Probably just a documentation thing.

commit 0c00f716 (2020-03-23): Fix #531, Doc MID use for CFE_TBL_NotifyByMessage

Documentation/comments only
Recommends unique MID for inter-app communications
closed 2020-03-27 by astrogeco