nasa/cFE #548: APIs Missing Unit Tests

dmknutsen, 2020-03-10:
**Describe the bug**
The following APIs need to have additional Unit Tests added:

cfe_es_api.c:CFE_ES_DeleteApp - Should test on boundary conditions for AppID
cfe_es_api.c:CFE_ES_GetAppIDByName - Don’t see where this is ever called in unit test
cfe_es_api.c:CFE_ES_ReloadApp - Should test on boundary conditions for AppID
cfe_es_api.c:CFE_ES_RestartApp - No success path test
cfe_esmempool.c:CFE_ES_GetMemPoolStats - No success path test
cfe_sb.h:CFE_SB_MsgId_Equal - Don’t see where this is ever called in unit test
cfe_sb.h:CFE_SB_MsgIdToValue - Don’t see where this is ever called in unit test
cfe_sb.h:CFE_SB_ValueToMsgId - Don’t see where this is ever called in unit test
cfe_sb_api.c:CFE_SB_SubscribeFull - There are 3 cases where   CFE_SB_BAD_ARGUMENT can get returned…UT should test each one individually.  Also this is outside the scope of the argument validation audit, but there are 3 additional return codes not tested in the unit test.
cfe_sb_api.c:CFE_SB_SubscribeLocal - Consider subscribing to message with limit greater than   CFE_PLATFORM_SB_DEFAULT_MSG_LIMIT
cfe_tbl_api.c:CFE_TBL_DumpToBuffer - No success path test
cfe_tbl_api.c:CFE_TBL_GetAddresses - Should test on boundary conditions for NumTables
cfe_tbl_api.c:CFE_TBL_GetStatus - Don’t see a success path test
cfe_tbl_api.c:CFE_TBL_ReleaseAddresses - No success path test. Also, should test on boundary conditions for NumTables
cfe_tbl_api.c:CFE_TBL_Update - Don’t see a success path test
cfe_tbl_api.c:CFE_TBL_Validate - No success path test
cfe_time_tone.c:CFE_TIME_Local1HzISR - Don’t see where this is ever called in unit test

**Expected behavior**
Unit Tests test all success/error paths

**System observed on:**
 NA code review/audit

**Reporter Info**
Dan Knutsen
NASA Goddard

ArielSAdamsNASA, 2021-08-10:
**1. cfe_es_api.c:CFE_ES_DeleteApp - Should test on boundary conditions for AppID**

In cFE/modules/es/fsw/src/cfe_es_api.c
- Checks if AppID is null
- Checks if AppID is an external cFE App
- Checks if AppID is not running 
- Does not check if App ID is greater than or less than the min/max values of possible APIDs, but checks if AppID is not CFE_ES_AppType_CORE.

In cFE/modules/es/ut-coverage/es_UT.c
- Test deleting an app that doesn't exist
- Does not check if App ID is greater than or less than the min/max values of possible APIDs

**2. cfe_es_api.c:CFE_ES_GetAppIDByName - Don’t see where this is ever called in unit test**

Called in cFE/modules/es/ut-coverage/es_UT.c

```
    /* Test CFE_ES_GetAppIDByName error with null AppID pointer and valid name */
    ES_ResetUnitTest();
    UtAssert_INT32_EQ(CFE_ES_GetAppIDByName(NULL, "UT"), CFE_ES_BAD_ARGUMENT);

    /* Test CFE_ES_GetAppIDByName error with valid AppID and NULL name */
    ES_ResetUnitTest();
    ES_UT_SetupSingleAppId(CFE_ES_AppType_EXTERNAL, CFE_ES_AppState_RUNNING, "UT", NULL, NULL);
    UtAssert_INT32_EQ(CFE_ES_GetAppIDByName(&AppId, NULL), CFE_ES_BAD_ARGUMENT);
```

**3. cfe_es_api.c:CFE_ES_ReloadApp - Should test on boundary conditions for AppID**

In cFE/modules/es/fsw/src/cfe_es_api.c
- Checks if AppID is null
- Checks if AppID is external 
- Checks if AppID is not running 
- Does not check if App ID is greater than or less than the min/max values of possible APIDs, but checks if AppID is not CFE_ES_AppType_CORE. 

In cFE/modules/es/ut-coverage/es_UT.c
- Test CFE_ES_ReloadApp with bad AppID argument
- Test reloading a core app
- Test reloading an app that is currently not running
- Test success initiating an app reload
- Test Reload app: file doesn't exist
- Does not check if App ID is greater than or less than the min/max values of possible APIDs, but checks if app is CFE_ES_AppType_CORE, CFE_ES_APPID_UNDEFINED, or CFE_ES_AppType_EXTERNAL.

**4. cfe_es_api.c:CFE_ES_RestartApp - No success path test**
- No success test in cFE/modules/es/ut-coverage/es_UT.c

**5. cfe_esmempool.c:CFE_ES_GetMemPoolStats - No success path test**
- No success test in cFE/modules/es/ut-coverage/es_UT.c

**6. cfe_sb.h:CFE_SB_MsgId_Equal - Don’t see where this is ever called in unit test**
- Called in cFE/modules/evs/ut-coverage/evs_UT.c
```
    /* Note implementation initializes both short and long message */
    UtAssert_INT32_EQ(UT_GetStubCount(UT_KEY(CFE_MSG_Init)), 2);
    UtAssert_INT32_EQ(UT_GetStubCount(UT_KEY(CFE_SB_TransmitMsg)), 1);
    CFE_UtAssert_TRUE(CFE_SB_MsgId_Equal(MsgData.MsgId, ShortFmtSnapshotData.MsgId));
    CFE_UtAssert_FALSE(CFE_SB_MsgId_Equal(MsgData.MsgId, LongFmtSnapshotData.MsgId));
```

**7. cfe_sb.h:CFE_SB_MsgIdToValue - Don’t see where this is ever called in unit test**
- Called in cFE/modules/msg/ut-coverage/test_cfe_msg_msgid_v1.c
```
    UtPrintf("Set msg to all 0, set msgid to max and verify");
    memset(&msg, 0, sizeof(msg));
    CFE_UtAssert_SUCCESS(CFE_MSG_GetMsgId(&msg, &msgid));
    UtAssert_INT32_EQ(CFE_SB_MsgIdToValue(msgid), 0);
    CFE_UtAssert_SUCCESS(CFE_MSG_SetMsgId(&msg, CFE_PLATFORM_SB_HIGHEST_VALID_MSGID));
    UT_DisplayPkt(&msg, sizeof(msg));
    CFE_UtAssert_SUCCESS(CFE_MSG_GetMsgId(&msg, &msgid));
    UtAssert_INT32_EQ(CFE_SB_MsgIdToValue(msgid), CFE_PLATFORM_SB_HIGHEST_VALID_MSGID);
```

**8. cfe_sb.h:CFE_SB_ValueToMsgId - Don’t see where this is ever called in unit test**
- Called in cFE/modules/msg/ut-coverage/test_cfe_msg_msgid_v1.c
```
    UtPrintf("Set ApId msgid bits only and verify");
    CFE_UtAssert_SUCCESS(CFE_MSG_SetMsgId(&msg, CFE_SB_ValueToMsgId(TEST_MAX_APID)));
    CFE_UtAssert_SUCCESS(CFE_MSG_GetMsgId(&msg, &msgid));
    UtAssert_INT32_EQ(Test_MSG_NotZero(&msg), MSG_APID_FLAG);
    CFE_UtAssert_SUCCESS(CFE_MSG_GetApId(&msg, &apid));
    UtAssert_INT32_EQ(apid, TEST_MAX_APID);
```

**9. cfe_sb_api.c:CFE_SB_SubscribeFull - There are 3 cases where CFE_SB_BAD_ARGUMENT can get returned…UT should test each one individually. Also this is outside the scope of the argument validation audit, but there are 3 additional return codes not tested in the unit test.**

**10. cfe_sb_api.c:CFE_SB_SubscribeLocal - Consider subscribing to message with limit greater than CFE_PLATFORM_SB_DEFAULT_MSG_LIMIT**

- Doesn't use CFE_PLATFORM_SB_DEFAULT_MSG_LIMIT, uses MsgLim instead. Does not use message greater than MsgLim. 

**11. cfe_tbl_api.c:CFE_TBL_DumpToBuffer - No success path test**
- No success test found in modules/tbl/ut-coverage/tbl_UT.c

**12. cfe_tbl_api.c:CFE_TBL_GetAddresses - Should test on boundary conditions for NumTables**
- No boundary conditions for NumTables in modules/tbl/fsw/src/cfe_tbl_api.c

**13. cfe_tbl_api.c:CFE_TBL_GetStatus - Don’t see a success path test**
- No success test in modules/tbl/ut-coverage/tbl_UT.c

**14. cfe_tbl_api.c:CFE_TBL_ReleaseAddresses - No success path test. Also, should test on boundary conditions for NumTables**
- No success test in modules/tbl/ut-coverage/tbl_UT.c
- No boundary conditions for NumTables in modules/tbl/fsw/src/cfe_tbl_api.c

**15. cfe_tbl_api.c:CFE_TBL_Update - Don’t see a success path test**
- No success test in modules/tbl/ut-coverage/tbl_UT.c

**16. cfe_tbl_api.c:CFE_TBL_Validate - No success path test**
- No success test in modules/tbl/ut-coverage/tbl_UT.c

**17. cfe_time_tone.c:CFE_TIME_Local1HzISR - Don’t see where this is ever called in unit test**
- Not called in files in the ut-coverage folder

skliper, 2021-08-11:
Great work.  ID checks against undefined along with a success case are sufficient for testing the "range" so those are good enough.  Confirming CFE_TIME_Local1HzISR will get called from the TIME coverage test update issue #473.  With this analysis and the remaining missing tests having specific issues, closing as duplicate/OBE.

closed 2021-08-11 by skliper