nasa/cFE #550: CFE cmake rules for tables copy the files into /cf, not /cf/tables where expected

MauriceS-gti, 2020-03-11:
**Describe the bug**
cFE cmake rules copy the *.tbl files into /cf/ and not into /cf/apps where most apps seem to expect them:

**To Reproduce**
Steps to reproduce the behavior:
1. Build your own cFS tree based on https://github.com/nasa/cFS
2. remove the *_lab apps
3. add the official repos for CFS_SCH. CFS_TO, CFS_TO, HK, HS and others
4. Configure everything
5. Build (I was able to build specifically 32 bit on my 64 bit machine using cmake 3.13 or newer - with the attached config)
[cFS_32bit_on_64bit.tar.gz](https://github.com/nasa/cFE/files/4319379/cFS_32bit_on_64bit.tar.gz)
6. run
```
.
.
1980-012-14:03:20.25884 ES Startup: SCH loaded and created
1980-012-14:03:20.25917 CFE_TBL:Load-App(5) Fail to load Tbl 'SCH.SCHED_DEF' from '/cf/apps/sch_def_schtbl.tbl' (Stat=0xFFFFFFFF)
EVS Port1 42/1/CFE_TBL 93: SCH Failed to Load 'SCH.SCHED_DEF' from '/cf/apps/sch_def_schtbl.tbl', Status=0xFFFFFFFF
EVS Port1 42/1/SCH 9: Error (RC=0xFFFFFFFF) Loading SDT with /cf/apps/sch_def_schtbl.tbl
.
.
```
7. Stop it, and do an ls
```
cpu1$ ls cf/*.tbl
cf/beer_lab_iloads.tbl  cf/gti_to_config.tbl  cf/hk_cpy_tbl.tbl  cf/lc_def_adt.tbl  cf/lc_def_wdt.tbl  cf/sch_def_msgtbl.tbl  cf/sch_def_schtbl.tbl

```
8. grep the app tree
```
~/source/cFS$ grep -r apps apps/ | grep "#define" | grep -v "\.html"
apps/hk/fsw/platform_inc/hk_platform_cfg.h:#define HK_COPY_TABLE_FILENAME  "/cf/apps/hk_cpy_tbl.tbl"
apps/sch/fsw/platform_inc/sch_platform_cfg.h:#define SCH_SCHEDULE_FILENAME  "/cf/apps/sch_def_schtbl.tbl"
apps/sch/fsw/platform_inc/sch_platform_cfg.h:#define SCH_MESSAGE_FILENAME   "/cf/apps/sch_def_msgtbl.tbl"
apps/md/fsw/platform_inc/md_platform_cfg.h:#define MD_TBL_FILENAME_FORMAT                "/cf/apps/md_dw%02d.tbl"
apps/md/fsw/unit_test/md_platform_cfg.h:#define MD_TBL_FILENAME_FORMAT                "/cf/apps/md_dw\%02d.tbl"
apps/fm/fsw/platform_inc/fm_platform_cfg.h:#define FM_TABLE_DEF_NAME               "/cf/apps/fm_freespace.tbl"
apps/sc/fsw/platform_inc/sc_platform_cfg.h:#define SC_ATS_FILE_NAME                "/cf/apps/sc_ats"
apps/sc/fsw/platform_inc/sc_platform_cfg.h:#define SC_APPEND_FILE_NAME             "/cf/apps/sc_append.tbl"
apps/sc/fsw/platform_inc/sc_platform_cfg.h:#define SC_RTS_FILE_NAME               "/cf/apps/sc_rts"
apps/sc/fsw/unit_test/sc_platform_cfg.h:#define SC_ATS_FILE_NAME                "/cf/apps/sc_ats"
apps/sc/fsw/unit_test/sc_platform_cfg.h:#define SC_APPEND_FILE_NAME             "/cf/apps/sc_append.tbl"
apps/sc/fsw/unit_test/sc_platform_cfg.h:#define SC_RTS_FILE_NAME               "/cf/apps/sc_rts"
apps/cs/fsw/platform_inc/cs_platform_cfg.h:#define CS_DEF_EEPROM_TABLE_FILENAME                "/cf/apps/cs_eepromtbl.tbl"
apps/cs/fsw/platform_inc/cs_platform_cfg.h:#define CS_DEF_MEMORY_TABLE_FILENAME                "/cf/apps/cs_memorytbl.tbl"
apps/cs/fsw/platform_inc/cs_platform_cfg.h:#define CS_DEF_TABLES_TABLE_FILENAME                "/cf/apps/cs_tablestbl.tbl"
apps/cs/fsw/platform_inc/cs_platform_cfg.h:#define CS_DEF_APP_TABLE_FILENAME                   "/cf/apps/cs_apptbl.tbl"
apps/hs/fsw/platform_inc/hs_platform_cfg.h:#define HS_AMT_FILENAME                "/cf/apps/hs_amt.tbl"
apps/hs/fsw/platform_inc/hs_platform_cfg.h:#define HS_EMT_FILENAME                "/cf/apps/hs_emt.tbl"
apps/hs/fsw/platform_inc/hs_platform_cfg.h:#define HS_XCT_FILENAME                "/cf/apps/hs_xct.tbl"
apps/hs/fsw/platform_inc/hs_platform_cfg.h:#define HS_MAT_FILENAME                "/cf/apps/hs_mat.tbl"
apps/hs/fsw/unit_test/hs_platform_cfg.h:#define HS_AMT_FILENAME                "/cf/apps/hs_amt.tbl"
apps/hs/fsw/unit_test/hs_platform_cfg.h:#define HS_EMT_FILENAME                "/cf/apps/hs_emt.tbl"
apps/hs/fsw/unit_test/hs_platform_cfg.h:#define HS_XCT_FILENAME                "/cf/apps/hs_xct.tbl"
apps/hs/fsw/unit_test/hs_platform_cfg.h:#define HS_MAT_FILENAME                "/cf/apps/hs_mat.tbl"
apps/beer_lab/fsw/tables/beer_lab_tbldefs.h:#define BEER_LAB_ILOAD_FILENAME     "/cf/apps/beer_lab_iloads.tbl"
apps/gti_to/fsw/src/gti_to_app.h:#define TO_CONFIG_FILENAME      "/cf/apps/to_config.tbl"
apps/gti_to/fsw/examples/multi_tf/to_platform_cfg.h:#define TO_CONFIG_FILENAME "/cf/apps/to_config.tbl"
apps/gti_to/fsw/examples/multi/to_platform_cfg.h:#define TO_CONFIG_FILENAME "/cf/apps/to_config.tbl"
apps/gti_to/fsw/examples/rs422/to_platform_cfg.h:#define TO_CONFIG_FILENAME "/cf/apps/to_config.tbl"
apps/ds/fsw/platform_inc/ds_platform_cfg.h:#define DS_DEF_DEST_FILENAME            "/cf/apps/ds_file_tbl.tbl"
apps/ds/fsw/platform_inc/ds_platform_cfg.h:#define DS_DEF_FILTER_FILENAME          "/cf/apps/ds_filter_tbl.tbl"
apps/ds/fsw/unit_test/ds_platform_cfg.h:#define DS_DEF_DEST_FILENAME            "/cf/apps/ds_file_tbl.tbl"
apps/ds/fsw/unit_test/ds_platform_cfg.h:#define DS_DEF_FILTER_FILENAME          "/cf/apps/ds_filter_tbl.tbl"
apps/lc/fsw/platform_inc/lc_platform_cfg.h:#define LC_WDT_FILENAME                "/cf/apps/lc_def_wdt.tbl"
apps/lc/fsw/platform_inc/lc_platform_cfg.h:#define LC_ADT_FILENAME                "/cf/apps/lc_def_adt.tbl"
apps/lc/fsw/unit_test/lc_platform_cfg.h:#define LC_WDT_FILENAME                "/cf/apps/lc_def_wdt.tbl"
apps/lc/fsw/unit_test/lc_platform_cfg.h:#define LC_ADT_FILENAME                "/cf/apps/lc_def_adt.tbl"
```
**Expected behavior**
All .TBL files should be installed into /cf/apps/

Change to **CFE**: cmake/arch_build.cmake:
install directory in add_cfe_tables needs to include apps
```
      # Create the install targets for all the tables
      install(FILES ${TABLE_DESTDIR}/${TBLWE}.tbl DESTINATION ${TGT}/${INSTALL_SUBDIR}apps)
    endforeach(TGT ${APP_INSTALL_LIST})
``` 

**System observed on:**
 - HW: Dell Laptop XPS 15
 - OS: Linux GTI-UID0110 5.3.0-26-generic #28~18.04.1-Ubuntu SMP Wed Dec 18 16:40:14 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux (Distro Linux Mint 19.3)
 - Versions [Versions:cFE 6.7.7.0, OSAL 5.0.6.0, PSP 1.4.4.0, chksm 60524 on Linux, SCH 2.2.2, CFS_TO commit 4589edb, CFS_CI, and other apps

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Maurice Smulders
Geneva Technologies Inc
255 Jefferson St, Ste 104
Monument, CO 80132

jphickey, 2020-03-11:
The `INSTALL_SUBDIR` is actually a function of the PSP and how one wants to stage files for the target.  There isn't one correct answer.

If you prefer to have an `apps` subdirectory, and you are using the `pc-linux` psp for simulation, then I suggest patching `psp/fsw/pc-linux/make/build-options.cmake` and changing the existing line to become:

`set(INSTALL_SUBDIR "cf/apps")`

Unfortunately this is an unconditional "set" and so it will override a previous setting (e.g. in your arch_build_custom.cmake file).  It might be advantageous to make it conditionally set, so a mission could override it without changing the PSP.

For the default framework config it was decided years ago to prefer a flat install tree, hence why everything goes in the single directory.

MSLaaf, 2020-03-12:
I understand, however the defaults of the separate packages as shown are in /cf/apps/, I did not change them from their Git repo version

jphickey, 2020-03-12:
Yes, agreed/understood there is a mismatch between the default config for apps and the default config for the framework.

It can be solved either way though (i.e. by removing `/apps/` in the app config(s), or by adding `/apps/` to your `INSTALL_SUBDIR`).

My point was that the CFE framework did make an intentional decision to default to a flat install directory for simplicity.  The issue is that nothing else followed suit, so maybe its worth re-discussing and re-considering that choice.

Either way, the install subdirectory is still a mission/platform choice, so we cannot hardcode an "apps" path into the common `arch_build.cmake` code, although `arch_build_custom.cmake` would be OK.

skliper, 2020-03-16:
@ejtimmon It would be nice if the apps worked "out of the box", are tickets already in to address this at the app level?  Or did this intentional decision not attempt to address breaking things down the line?

ejtimmon, 2020-03-16:
There are tickets to address this for all of the apps (just replacing "/cf/apps" with "/cf" in paths to files and tables)

skliper, 2020-03-16:
Excellent!  Unfortunately due to the different release processes and the out of sync development/release cycles apps and the framework/bundle master are not always self compatible.  The goal is to eventually report what version of the framework each app was last tested against or vice versa (apps that run against a version of the cFS framework).  See Compatible list of cFS apps on the front page of the bundle.

closed 2020-03-16 by skliper