nasa/cFE #552: Remove undefined CFE_OS_ERROR_TASK_ID in cfe_error.h

skliper, 2020-03-11:
**Is your feature request related to a problem? Please describe.**
https://github.com/nasa/cFE/blob/f1be0482d7a9493cb35613a8e43020a66a4a64d3/fsw/cfe-core/src/inc/cfe_error.h#L757

and there is no OS_ERROR_TASK_ID

**Describe the solution you'd like**
Delete

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-03-12:
Actually, I would propose removing _all_ of the `CFE_OS_` error codes.  They only occur in one place in the unit test and that place is wrong (should be using the OSAL-provided constant).

There should be no need for these duplicate names for the same error codes.  It would be a different story if they were providing some abstraction, but they are not- they are only providing a different name for the same thing.

skliper, 2020-03-12:
Yeah, I thought this topic had been touched on recently but I didn't find an issue (maybe in comments somewhere?)  I'd be interested to hear from @acudmore.

jphickey, 2020-03-12:
It came up in the review of a recent change (issue #308/pull #452) where it was initially mixing the `OS_ERR_` codes (direct from osal) and these `CFE_OS_` synonyms.  That changeset was subsequently modified to use only the OSAL error codes.

Logically speaking - If calling an OSAL function directly, it should be comparing to the OSAL error codes directly too.

The gray area comes in when CFE API calls in turn call the OSAL API, and return the OSAL error directly.  For true abstraction this is a bad paradigm and shouldn't even be done (it should explicitly translate the code).  But for efficiency the CFE error codes are simply chosen to not overlap/alias OSAL codes to make this practice OK, and it is understood that a return code from a CFE API might be either an OSAL error code or a CFE error code.  Since both sets agree on the general pattern that negative is error and non-negative is success, its fine.

At any rate, `cfe_error.h` internally also includes `osapi.h` so if an application does a `#include <cfe_error.h>` then this will provide the OSAL error codes too.

So long as we aren't explicitly translating OSAL error codes to CFE error codes, we do not need to redefine the OSAL error constants in CFE either (and should not) .

skliper, 2020-03-12:
Bigger issue probably worth 2 steps - deprecate then remove, the undefined referenced here could just be removed (they'd likely already be broken).

skliper, 2020-04-02:
See #580 to cover deprecation of the entire set.

PR #582 (2020-04-21): Fix #580, Deprecate CFE_OS_ abstracted error codes
**Describe the contribution**
Fix #580 - deprecates CFE_OS_ abstracted OS error codes
Added CFE_OMIT_DEPRECATED_6_7 just to be consistent
Also fix #552 - removes non-existent codes

**Testing performed**
Steps taken to test the contribution:
1. CI - see https://travis-ci.com/github/skliper/cFS/builds/157872188

**Expected behavior changes**
None

**System(s) tested on**
 - Hardware: AMD
 - OS: Ubuntu 18.04
 - Versions: bundle + this change

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED fo [... cut]
commit fe700b7f (2020-04-21): Fix #580, Deprecate CFE_OS_ abstracted error codes

Added CFE_OMIT_DEPRECATED_6_7 just to be consistent
Also fix #552 - removes non-existent codes
closed 2020-04-27 by astrogeco