nasa/cFE #556: Document deprecated elements

skliper, 2020-03-13:
Looks like template broke... anyways:

Add \deprecated and \name the group where appropriate relative to the user's guide documentation such that it shows up as deprecated.

Jacob Hageman - NASA/GSFC

skliper, 2020-03-13:
Suggest commenting the code typically provides a replacement or alternative where appropriate.

skliper, 2020-03-24:
\name didn't work so well, follow doxygen deprecation pattern established in OSAL.  Suggest adding "DEPRECATED" at the start of the brief for every element that is deprecated (so it's obvious even if just the brief is shown).

skliper, 2020-05-12:
@lbleier-GSFC - did you intend to reference this issue from your commits?  Or #559?  Or does it fix both (if so it should reference both)

lbleier-GSFC, 2020-05-12:
> @lbleier-GSFC - did you intend to reference this issue from your commits? Or #559? Or does it fix both (if so it should reference both)

Whoops. This was supposed to be #559

skliper, 2021-01-11:
OBE/Duplicated of other deprecation work. Any newly deprecated elements should include @deprecated flag.

closed 2021-01-11 by skliper