nasa/cFE #564: CFE_EVS event filtering doesn't seem to work correctly

tngo67, 2020-03-18:
**Describe the bug**
EVS doesn't seem to filter out the mask for CFE_EVS_FIRST_ONE_STOP.  I have not tried other pre-defined filter mask values.

**To Reproduce**
Steps to reproduce the behavior:
1. Build & run SCH as-is with just CI & TO as part of the FSW.  Make sure TO subscribes to short/long event messages.  Increase TO TLM pipe depth to max depth.
2. Observe the console output from FSW.  TO TLM pipe would overflow with SCH as sender

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware
 - OS: CentOS 6.10 & 7.x
 - Versions: cFE 6..x7, OSAL 5.0.x

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable

skliper, 2020-03-18:
@dmknutsen - could you look into this one?  Should be able to test by just setting the filter on any app/service and send a couple noops?

dmknutsen, 2020-03-18:
Yep

dmknutsen, 2020-03-23:
I tested filtering via the mask CFE_EVS_FIRST_ONE_STOP (and spot checked the others) by executing the following test:

1. Started CFE
2. Launched GroundSystem.py
3. Sent Command: CFE_EVS_APP_EVENT_FILTER
       - AppName: CFE_SB
       - EventID : 28
       - Mask: 0
4. Send Command: CFE_EVS_SET_FILTER
       - AppName: CFE_SB
       - EventID : 28
       - Mask: 0xFFFF
5. Sent Command: CFE_SB_NOOP_CC
6. Verified via console output SB NOOP was recieved
7. Sent Command: CFE_SB_NOOP_CC
       - Repeated this step several times and verified that additional NOOPs were being filtered
8. Repeated this test for various other EVS filters to ensure that filter masks are working as expected

As far as I can tell, filtering seems to be working correctly.  It might be possible that there is an issue with the way the filter(s) are being implemented in your app.  Could you post your code and/or provide additional steps to reproduce the issue?

Thanks,

Dan

tngo67, 2020-03-23:
Please take a look at how event filters are initialized for SCH app (github.com/nasa/sch).  When we build & run SCH, the events don't seem to get filtered.  You don't need to send command to filter the events since the events are initialized with CFE_EVS_FIRST_ONE_STOP filter mask value.

dmknutsen, 2020-03-23:
Will do.  Thanks for the info.

skliper, 2020-03-23:
@tngo67 Is there a call to CFE_EVS_ResetAllFilters() in your SCH_HousekeepingCmd?

This could be your problem.  Not sure what the intent was with this call in your housekeeping command handler.

skliper, 2020-03-23:
Just found it in the GSFC version also (maybe they are the same?) @ejtimmon Do you know why scheduler resets all filters in HK command processing?

tngo67, 2020-03-23:
Ah.  I think you have something there.  That would explain it.  Thanks!

ejtimmon, 2020-03-23:
No, I don't know.  I'll look into and get back to you.

ejtimmon, 2020-03-24:
I don’t see any discussion of the filtering behavior in the SCH documentation.  The SCH filter includes events like SCH_SAME_SLOT_EID, SCH_MULTI_SLOTS_EID, and SCH_SKIPPED_SLOTS_EID – events that could potentially be sent many times during one iteration of the schedule.  My theory is that the filter is in place so that you see these events only once during a single iteration of the schedule, but the filter is reset so that you can see them once for each iteration.  For instance, if slots are being skipped, you don’t need (or want) 50 separate notifications for one schedule iteration, but you do need to know if that behavior is persisting across multiple iterations of the schedule.  I think that the filter reset happens in the housekeeping command because theoretically, that command should only be getting serviced after a schedule iteration has been completed.

jwilmot, 2020-03-24:
I would suggest that the filter be set to 1 out of 8 or some other number and remove the command. Use the counter to tell you how often/many. The idea was you fix the issue then reset the filter counter.  The reset filter command violated the undocumented but expected design pattern and looked like a bug.

skliper, 2020-03-24:
Or update the design such that it sends the related event a maximum of once per iteration of the schedule when unfiltered with an associated count and whatever information is appropriate for the last occurrence of that type of event.  Then the behavior becomes less dependent on different schedules, less of a event spewer without requiring messing with filters.

Then behavior is close to the same for low-rate events, and naturally limits itself at the high end.

ejtimmon, 2020-03-24:
I've created an SCH ticket to address this.

astrogeco, 2020-03-27:
@skliper and @ejtimmon should we close this issue and mark it as "gsfc-apps" or something like that?

skliper, 2020-03-27:
I'm fine with just closing as a resolved question (relating to cFE).

closed 2020-03-27 by skliper