nasa/cFE #567: ExecutionCounter should be associated with an app, not a task.

jphickey, 2020-03-23:
**Is your feature request related to a problem? Please describe.**

The "ExecutionCounter" metric is currently stored within the Task record, inside the `CFE_ES_TaskRecord_t` structure.

However, only an "app" really has the notion of an execution counter, which is the number of times that app called the `CFE_ES_RunLoop` API.

Furthermore, inside the `CFE_ES_RunLoop()` function itself, it looks up the MainTaskId of the caller and only increments _that_ execution counter, not the execution counter of its own task record.  This means no matter what child task actually calls this function, it implements the execution counter of the main task only.

**Describe the solution you'd like**
The `ExecutionCounter` metric should be stored and maintained within the app record, not the task record.  This would simplify things in a number of places, and be more consistent with how it is actually used.

**Additional context**
Noticed this when fixing #480 and it seemed rather odd/incorrect to be storing the ExecutionCounter where it is.  This causes the code to jump to other entries in the table for the sole purpose of reading/updating this value, when it already had the correct app record to start.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

ejtimmon, 2020-03-23:
FWIW I recently got some questions from app developers about this (wanting to know whether the exec counter should be incremented in child tasks) - the change suggested would clarify that.  I did get the sense from those users that they wanted a way to track the execution of child tasks in addition to the main task.

jphickey, 2020-03-24:
@ejtimmon After initially writing this issue I then rediscovered that there is a function called `CFE_ES_IncrementTaskCounter` which increments the counter for the task context which invoked it.

So yes, there is actually the notion of a child task counter....

But then it is still unclear to me why the `CFE_ES_RunLoop` function would increment the counter for  the _main_ task of the app, rather than the task that actually invoked the function - this doesn't make sense, because if the main task also calls `CFE_ES_IncrementTaskCounter()` like child tasks do then it will get double incremented, or if a child task calls `CFE_ES_RunLoop()` then that activity will actually get accounted for in the main task (which is just plain wrong).

I suggest one of the following:

1. Just have `CFE_ES_RunLoop()` invoke `CFE_ES_IncrementTaskCounter()` to increment the counter for the task from which it was called.  So if it gets called from a child task, then that child task gets incremented, not the main task.
2. Maintain a separate "app" exec counter which is incremented by `CFE_ES_RunLoop` (only), and use the `CFE_ES_IncrementTaskCounter` for regular task activity accounting.

At the very minimum, it is probably at least worth a note in the documentation of CFE_ES_RunLoop that it is incrementing a task counter (whichever one it is) because the documentation doesn't seem to mention it at all, and this could certainly confuse someone (it confused me...).

Probably worth a quick discussion to see what others think...

ejtimmon, 2020-03-24:
Whenever a decision is made, we should also add guidance to the App Developer's Guide.

astrogeco, 2020-03-25:
CCB 2020-03-25: Discussed, will have telemetry impacts based on app vs task counter. Will also affect Health and Safety's behavior. Behavior needs to be documented in users guide.

skliper, 2021-01-11:
Is this needed for Caelum?  Should we touch on again in CCB?

jphickey, 2021-01-11:
Note that previous changes in this area _did_ in fact change `CFE_ES_RunLoop` to call the `CFE_ES_IncrementTaskCounter` to increment the counter.  Simply put - it has to be this way because having any task increment a counter other than its own (i.e. the old way) would be a race condition.  So this is not only simpler - just one common function to increment exec counter - it is also safer.

If guidelines/templates were followed this is no effective change because `CFE_ES_RunLoop` is called from the main task - so its the same counter.  But if guidelines/templates were not followed and this is called from a child task then it will increment the counter of the child task, not the main task.

jphickey, 2021-01-11:
If anything we should just update the documentation of `CFE_ES_RunLoop` to indicate this -- that it increments the task execution counter of the task that called it, and is expected to be invoked from the "main" task of a CFE application, as part of that app's main loop.

skliper, 2021-01-11:
Implementation change addressed by #598.  Remaining work here is to update documentation.

PR #1601 (2021-06-10): Fix #567, Document CFE_ES_RunLoop increment task counter behavior
**Describe the contribution**
Fix #567 - updated documentation

**Testing performed**
CI - documentation only

**Expected behavior changes**
None

**System(s) tested on**
CI

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
commit 11776a38 (2021-06-10): Fix #567, Document CFE_ES_RunLoop increment task counter behavior
commit 6ffe2c0d (2021-06-10): Merge pull request #1601 from skliper/fis567-runloop_inc_count

Fix #567, Document CFE_ES_RunLoop increment task counter behavior
PR #1619 (2021-06-16): cFE Integration candidate: 2021-06-15
## Testing

cFE Checks <https://github.com/nasa/cFE/pull/1619/checks> 
- Documentation workflow failures due to osal docs rename. These are fixed in bundle.
- Local tests failure due to osal docs rename, also fixed in bundle. 

cFS Bundle Checks <https://github.com/nasa/cFS/pull/265/checks> 

## Context

Part of https://github.com/nasa/cFS/pull/265

## Authors

@jphickey 
@zanzaben 
@skliper 
@pepepr08 

## Description

### PR #1590 
Fix #1483, Add CI workflow to run cFE coverage tests

### PR #1598 
Fix #1299, Updated FS Read/WriteHeader API return docume [... cut]
closed 2021-06-16 by astrogeco