nasa/cFE #576: unify "table" and ".scr" file loading

CDKnightNASA, 2020-03-31:
**Is your feature request related to a problem? Please describe.**
We currently have two ways of populating configuration data in cFS, the "table" API (loading a binary data structure into memory) and a "configuration file" (cfe_es_startup.scr). These have similar purposes (load-time configuration) and could use the same API?

**Describe the solution you'd like**
It should be fairly straightforward to automagically determine whether a "table" file is an ELF or a text file (first byte == 0x7F => ELF) and if it's not ELF it should hand off to a text parser (provided by OSAL?) to parse the table into an in-memory data structure. The current "csv-like" structure wouldn't handle complex (e.g. nested) data structures but for common tables it should be sufficient. Alternatives would be JSON or YAML or some other text format, but would require a way to disambiguate.

Table files trade efficiency for easy editing.

**Describe alternatives you've considered**
Stick with the separate ES code for loading the .scr file and table code. SBN previously used a similar .csv file format for its load configuration but it's been moved to a table-based configuration.

**Requester Info**
Christopher.D.Knight@nasa.gov

jphickey, 2020-04-01:
Actually, I'd be in favor of getting rid of the text files entirely.  Just make the startup script into a "table" that gets loaded by ES during startup.  This would get rid of a lot of code from ES that parses text files, really no need for all that on board.  Unless someone wants to edit their start up scripts in place?

CDKnightNASA, 2020-04-01:
I regularly find myself having to go in and tweak the startup script post "make". Text-based "tables" does lend itself to rapid prototyping and debugging.

ejtimmon, 2020-04-01:
I second that.  

Also, is it still true that cFE can be built without Table Services?  I don't know how many users actually do that, but making the startup script into a table would take away that option.

skliper, 2020-04-02:
I recently did test a build without table services as part of the documentation updates.  It still works.

jphickey, 2020-04-02:
Sounds like people are not in favor of removing the FSW text-based file parsing.... to me it always seemed odd to have one and only one file which was text-based, in a one-off (non-standard) syntax, with a custom parser.  That text parsing code has also been a source of bugs in the past.

A few other thoughts then:
- If we want text parsing on-board, we should embrace a standard format (YAML and JSON or even Lua syntax works well - the latter having the advantage of allowing actual control statements, at the expense of other 3rd party dependencies of course.
- But Whatever text format is settled upon, the parsing logic should at least be accessible and usable to apps through some API.  It really never made sense to me why all this logic to parse a text file exists, taking up code space forever for a one-time execution at startup, never to be used again until a reboot.  If we are expending the code space for all this logic, then at least offer it to be used in apps too.
 
Regarding TBL services:
- TBL services _is_ the recommended way of configuring apps.  Without this you'll have to hardcode everything.
- If TBL services is optional, why is it a core app and not something we load in like other CFS apps?

Final thoughts:
- My main gripe is really that we have two different loadable file formats in play for not really a good reason.  If we like binary for simplicity, then great, use binary, but if we like text for maintainability, lets use text for tables too (JSON/YAML/Lua syntax etc can all define structured data).
- Also note CFE TBL files are not ELF files, they are custom binary formats that start with the CFE FS header
- But I'm not sold on the need for dynamic identification logic -- the app doing the load should know what type of file to expect.

Sorry for posting more questions than answers about this....

CDKnightNASA, 2020-04-02:
I'll add to the pile of questions...Another option I thought of is a binary editor...I don't know if there are any open-source binary editors that support feeding in a struct definition but it should be possible to do so (Python's struct library, particularly, would be sensible.) This would allow for binary only but give the flexibility of post-build configuration changes.

Certainly if the CCB agrees no more text configs, I'm ok with that. I'm just on the fence at this point and am exploring options.

astrogeco, 2020-04-02:
I really like the standard text format idea. I would vote for YAML since they're easy on the eyes. I just started playing with Lua though and it also looks powerful and well thought out.

CDKnightNASA, 2020-04-02:
There are a number of C YAML libraries, all of which are under the MIT open source license...I'm assuming crossing the licensing streams is not as bad as crossing the proton pack streams.

acudmore, 2020-04-02:
**Regarding Text file format and parsing:**
I agree with @jphickey about having a standard format to parse, and making that API available to apps to use. Although that might encourage app developers to abandon binary tables in favor of text based tables.

**Regarding ES Startup Script being binary vs text:**
Making the cFE ES startup script a binary table (TBL services) would certainly simplify the startup code and unify how we deal with loadable parameter data. It would also allow us to verify the integrity of the file, which I don't think is being done now. But that would require the table services to be included, which leads to..

**Regarding optional table services:**
There is one major organization that does not use the cFE table services, so they would like to keep the table services as optional. Most of our cFS apps use tables, including the changes recently made to the bundle apps, so most of the users are depending on the table services. 

There was a recent thread on the community mailing list asking about how to configure an app for different CPUs. With the old build system, we could setup a platform include directory for each CPU, which would allow us to have different message IDs for different CPUs. It was suggested that we use tables for parameters such as message IDs, and I actually think that is a good solution. It would allow the same binary app to run on different CPUs with different tables.

**Ease of making table changes:**
One of the things that detracts from using tables is the pain of compiling the table and using the elf2cfetbl tool, but that is largely taken care of by the build system now. If the cfs startup script was a binary table, I don't think it would be too hard to make a quick edit and run the make, as opposed to just editing the the text file in place to make a quick change.
I think it would be great to have a table tool that can take a non C format such as JSON to build the binary table without going through the steps of compiling the table and converting elf to the binary format. ( plus that would open up the possibility of building tables on systems that don't have native ELF support ). Isn't that what EDS aims to provide?

**Final Thoughts:**
I am starting to prefer the idea of using a binary table for the cFE ES Startup script, and I really like the idea of putting the message IDs in a table for each app. Could we even consider putting the cFE Core message IDs in a loadable table?
This would make the table services non-optional, and hopefully that would not have much of an impact on organizations that do not use table services now. It would be used in the cFE core, but still would not be required for non GSFC cFS apps.

skliper, 2020-04-02:
> There are a number of C YAML libraries, all of which are under the MIT open source license...I'm assuming crossing the licensing streams is not as bad as crossing the proton pack streams.

Depends if we clone and own, or just link to it.  I think link is preferred from the open source license point of view, and deployments could always clone/own/fork if they want for security/maintenance issues.

skliper, 2020-04-02:
> I really like the standard text format idea. I would vote for YAML since they're easy on the eyes. I just started playing with Lua though and it also looks powerful and well thought out.

I vote STOL. (Just kidding, that's a knock on a big project we adore)

Actually I'd prefer JSON - "JSON is a lightweight format for storing and transporting data."  I'd be concerned with scope creep on the other languages...

jphickey, 2020-04-02:
Thanks @acudmore  for the thoughts!

I was under the impression that the impetus for running without CFE TBL was due to the memory footprint.  But has anyone put actual numbers for text/data usage of CFE TBL services?  How much are we actually saving by not including this subsystem at all, in comparison to the features we give up?  How does it compare to the amount of code in ES that does the one-off text file processing?  Maybe there is a way we could do a compromise "TBL-lite" by omitting the management commands and extra non-critical features, and combined with removing the ES text-processing code, the overall runtime footprint might be not much different?

**Regarding text-based table files:**
This is entirely possible to define your table data in YAML/JSON/Lua/Whatever if/when an EDS-type table definition is used.  The problem with parsing text on board though - as always - is memory for all the data definitions.  If we were at the point where some project removed TBL services to save memory, the footprint of having all those text-strings in memory to support on-board parsing of structured text files is significant.   

This is why even in the EDS work done by GHAPS the build still has a binary table file format for FSW loading.  But rather than `elf2cfetbl` it uses an alternative tool that parses a text file as part of the build and then generates a binary from that.  This avoids the ELF/compiler dependency but keeps the binary file model basically the same.

jphickey, 2020-04-02:
Also, I meant to comment on the MsgId aspect of this:

I still prefer to put any MsgId configurations in a table rather than hardcoding them.  But in case tbl services are not available or there is a reason that it won't work, the alternative is to subscribe by "TopicId" rather than subscribing by "MsgId". 

Simply put, the TopicId is a non-instance-specific descriptor of a certain stream (e.g. CFE_ES TLM).  By combining with an instance number at run time (e.g. CPU number) we can then lookup the actual MsgId to subscribe to.  This is a sort of middle-ground between going fully dynamic/table based and fully static/compile-time MsgId subscriptions.  Most importantly, it does generally allow the same binaries to be loaded on multiple CPU instances and it just works.

(don't mean to muddy the conversation too much, we might move this to the separate MsgId discussions that have been going on)

skliper, 2020-04-02:
> But has anyone put actual numbers for text/data usage of CFE TBL services?

Just did this with the user's guide updates.

https://github.com/nasa/cFE/blob/62252d11409f337d3dea2732739e068987363985/docs/src/cfe_tbl.dox#L31-L62

skliper, 2020-04-02:
And now that I read that there is definitely a typo... 144M...

acudmore, 2020-04-02:
I wonder if we can have TBL-Lite services built in to load cFE Core configuration and the startup script, then have the full table services as an app? Or if it stays in the cFE core, the full service can be compiled out for users that don't use the table services in their apps.

Just to summarize, I like the idea of going with binary tables for the startup script and even app configuration data such as message IDs. I also like the idea of subscribing to a topicID, I'm a fan of MQTT, but I'll leave that for the appropriate discussion. Having message IDs in a binary table is a step in the right direction. 

I also just saw the sizes that @skliper posted above. I don't think that is big enough to worry about. We even had tables on the Dellingr Cubesat with 2MB of SRAM.
For an organization that does not use the table services, having dead code/commands/APIs might be a bigger concern.

astrogeco, 2020-04-02:
@acudmore would you be strongly opposed to going the text-based as opposed to binary? What are the downsides other than file size?

astrogeco, 2020-04-02:
> I vote STOL. (Just kidding, that's a knock on a big project we adore)
>
> Actually I'd prefer JSON - "JSON is a lightweight format for storing and transporting data." I'd be concerned with scope creep [...]

If we ever want an on scripting language I think Lua might be the way to go: https://www.lua.org/about.html

The JSON vs YAML conversation boils down to purpose. JSON is human readable but not as edit-friendly as YAML. Also YAML is a superset of JSON so we kind of get two for one :)

skliper, 2020-04-02:
> > > I really like the standard text format idea. I would vote for YAML since they're easy on the eyes. I just started playing with Lua though and it also looks powerful and well thought out.
> > 
> > 
> > I vote STOL. (Just kidding, that's a knock on a big project we adore)
> > Actually I'd prefer JSON - "JSON is a lightweight format for storing and transporting data." I'd be concerned with scope creep on the other languages...
> 
> If we ever want an on-board scripting language I think Lua might be the way to go: https://www.lua.org/about.html
> 
> The JSON vs YAML conversation boils down to purpose. JSON is human readable but not as edit-friendly as YAML. Also YAML is a superset of JSON so we kind of get two for one :)

The topic is tables (data/parameter sets), not on-board scripting...

CDKnightNASA, 2020-04-02:
Sounds like the consensus is to remove the .scr and have ES use a table for configuration (and support the "lite" table configuration for deployments wanting a smaller footprint?) I'll close this ticket and create a new one for ES.

closed 2020-04-02 by CDKnightNASA

acudmore, 2020-04-02:
@astrogeco , I like the binary table approach for consistency and efficiency on the target system. I also think we can make the process of creating tables user friendly on the ground. No reason why we can't have JSON or YAML text files as the source as long as we can back it up with a simple conversion/build process. 

I don't have any evidence that using a binary file is safer or less error prone, but seems that it would be (less code, less string manipulation).

jphickey, 2020-04-02:
Wow, stepped away from email for a bit and missed a bunch of discussion here.

Final thoughts:
- I'm a big fan of Lua.  Its (relatively) small, fast, and highly flexible, and I think it fits the bill quite nicely for any FSW deployment that needs any sort of on-board scripting or automation.   I've even run a Lua interpreter as a CFS app and it works great.
- That being said, if all you want to do is parse a startup script, its overkill, and even with its small size is way bigger than a simple JSON/YAML parser.
- But if there was separately a need for onboard scripting, then _also_ using it to run a startup script makes perfect sense.

Regarding "unused commands" and "TBL-lite" concepts:
- I still recommend going full-modular with commands allowing EVERY ground command in the core apps to be individually enabled/disabled via compile-time configuration.  Disabled commands would not be compiled in.
- This topic has come up before with e.g. the CFE_ES shell function.  Some missions need it, whereas for some its a security risk.
- Allowing one to compile-out all of the dynamic TBL management commands but keeping TBL for internal use only like startup config might be a good alternative to compiling out TBL entirely (especially if we can clean up some ugly code in ES by doing this)

skliper, 2021-09-24:
Moved to #583