nasa/cFE #583: ES should use a table for the "startup script"

CDKnightNASA, 2020-04-02:
**Is your feature request related to a problem? Please describe.**
The ES "startup script" file (a CSV-like configuration of which applications and libraries to load at runtime) is not very well designed (using sscanf) and not consistent with how cFS generally manages runtime configurations with tables.

**Describe the solution you'd like**
Replace the .scr code with a table.

**Describe alternatives you've considered**
One alternative is to update the .scr parser, and/or use one/more industry-standard file formats (JSON, YAML) and an open-source parser that we would include (copy into) in our codebase. General consensus is that the CCB prefers going with the standard table mechanism.

**Additional context**
Might be interesting to see if there's a way to change a table without having to re-compile...Otherwise would be helpful to have a document detailing how, post deploy, an operator can update the ES configuration without having to do a full re-deploy.

**Requester Info**
Christopher.D.Knight@nasa.gov

CDKnightNASA, 2020-04-02:
ref issue #576

CDKnightNASA, 2020-04-28:
So it looks like the existing es_UT.c does not actually load/run any applications/libraries, I will assume the new es_UT.c does not need to either.

skliper, 2020-04-28:
As a coverage test all you need to cover is the unit under test... any actual load/run would be a functional test (requires full stack, not OSAL stubs).