nasa/cFE #594: Ability to register a function call for the performance manager (or call a PSP stub)

skliper, 2020-04-08:
**Is your feature request related to a problem? Please describe.**
Sounds like historical (and current) missions typically end up adding calls to hardware from the performance calls.

**Describe the solution you'd like**
Either add in a PSP API and call from ES, or support an API to add a function call.

**Describe alternatives you've considered**
None.

**Additional context**
None.

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-04-08:
The existing `CFE_PSP_Get_Timebase` API is used exclusively by the performance monitor code.  Couldn't whatever platform-specific logic be added as part of this?

That being said - there are two concerns I would have about doing that:
- it will only be called when performance log is actually active (most of the time it isn't, until commanded, and then it will auto-stop after trigger if set up that way).
- The call is made (or will be made) outside of any lock so it needs to reentrant/thread-safe, but shouldn't do any locking of its own.  so any ops would need to be fairly primitive.

skliper, 2020-04-08:
I assume they are just twiddling bits in a register so they can hook up a logic analyzer to an IO line, but would need to be worked with stakeholders.  Either way, not a current priority just a enhancement topic for future discussion.