nasa/cFE #596: Unit test - split "AND"-ed conditionals into separate asserts

jphickey, 2020-04-09:
**Is your feature request related to a problem? Please describe.**
Debugging unit tests can be very difficult, frustrating, and time consuming.  One major part of the problem which makes them very debugging-unfriendly is something like the following:

```
    UT_Report(__FILE__, __LINE__,
              UT_EventIsInHistory(CFE_ES_PCR_ERR2_EID) &&
              CFE_ES_Global.AppTable[Id].ControlReq.AppTimer == 0 &&
              CFE_ES_Global.AppTable[Id].ControlReq.AppControlRequest == CFE_ES_RunStatus_SYS_DELETE,
              "CFE_ES_ScanAppTable",
              "Waiting; process control request");
```

The problem with this type of construct is that there are 3 separate tests being combined into one single assert.  When it fails, it is not possible to see which of the three conditions are evaluating false.  Many of them call functions within the test case, too, which further obfuscate what the actual return value was.  The only way to test this is run it in a debugger, break at the start of the test, then set a breakpoint inside e.g. UT_EventIsInHistory to see what it returned.

**Describe the solution you'd like**

1. At a minimum - split the `&&` conditions into separate asserts.  This would _at least_ let the developer know which one is actually the fault.  
2. Nice to have - employ the macros similar to what @CDKnightNASA added in nasa/osal#405, which show the _values_ tested in the log, not simply just a pass/fail.

**Describe alternatives you've considered**
Continue struggling to figure out what actually went wrong every time a UT failure comes up.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2020-04-09:
Added milestone, targeted for this round for unit test consistency.

CDKnightNASA, 2020-04-09:
Now that the UT macros for OSAL have been accepted I can take a look at using that model for CFE as well (I have a crack at sb_UT.c but I like the OSAL macros even more so I will revisit.)

jphickey, 2021-06-16:
In order to cross reference/validate that the test cases are covering the documented functions, need to do a scrub of all "Report" outputs and make them use the proper macros that include full context info.

PR #1624 (2021-06-23): Fix #596, scrub all UT_Report calls
**Describe the contribution**
Scrub through all UT reporting calls and replace with preferred macro where possible.

- All calls to check status/return code are replaced with macro that logs all arguments and return value
- All calls that involved multiple conditions AND'ed together are replaced with individual asserts on each condition.

Fixes #596
Fixes #470 

**Testing performed**
Build and run unit tests, confirm logging is as intended, and coverage level is not reduced

**Expected behavior changes**
Better, more complete context information is logged in [... cut]
commit 0701923e (2021-06-23): Merge pull request #1624 from jphickey/fix-596-ut-report-macros

Fix #596, #470, scrub all UT_Report calls
commit 890cc87c (2021-06-23): Fix #596, remove now-unused assert functions (final)

Clean up the assert functions and macros which are no longer
used after updating all coverage tests to use preferred macros.
PR #1632 (2021-06-23): cFE Integration candidate: 2021-06-22
## Context

Part of https://github.com/nasa/cfs/pull/270

## Testing

cFE Checks <https://github.com/nasa/cFE/pull/1632/checks>
Bundle Checks <https://github.com/nasa/cfs/pull/270/checks>

## Authors 

@skliper 
@zanzaben 
@jphickey 
@pepepr08 

## Description

### PR #1627

Fix #1626, rename/clean CFE coverage assert macros

### PR #1630 

Fix #1629, correct path to users guide warning log

### PR #1621 

Fix #1477, add additional test cases for Child Tasks

### PR #1608 
Fix #1367, Add cfe functional tests to CI

### PR #1623 

Fix #465, Added UT tests f [... cut]
closed 2021-06-23 by astrogeco