nasa/cFE #599: CFE_EVS_GenerateEventTelemetry should check return code from CFE_SB_SendMsg()

CDKnightNASA, 2020-04-09:
**Is your feature request related to a problem? Please describe.**
EVS, when called to send events, eventually calls down to CFE_EVS_GenerateEventTelemetry() which sends the event to the ports and to the SB via CFE_SB_SendMsg(). However, it currently does not check the return code from SendMsg().

**Describe the solution you'd like**
GenerateEventTelemetry() should report an error to SysLog (and the ports?) if it's unable to send the event to the SB (preferably including the original event inside the SysLog error report.)

**Describe alternatives you've considered**
Leaving it as it is, it's unlikely that CFE_SB_SendMsg() would generate an error.

**Additional context**
Note also that if SendMsg() [really SendMsgFull()] generated an error, it would send an event through EVS, causing a loop between SB code and EVS code (that will likely go on until the system crashes). Again, this condition is unlikely, but this logic should be fixed to prevent loops.

**Requester Info**
Christopher.D.Knight@nasa.gov

jphickey, 2020-04-09:
Haven't looked at this particular case, but I think there is some sort of loop prevention logic in play here to prevent this sort of thing.

CDKnightNASA, 2020-04-10:
I ...suppose... this lock will block on the second time through the loop:

https://github.com/nasa/cFE/blob/60a5f6580d1e737f46efce0418aba5b8dac6df6a/fsw/cfe-core/src/sb/cfe_sb_api.c#L1233

closed 2020-04-10 by CDKnightNASA

skliper, 2021-09-24:
Duplicate of #1952