nasa/cFE #603: CFE_EVS_SendEvent stub should provide full context

gilliganeric, 2020-04-10:
**Describe the bug**
The CFE_EVS_SendEvent stub for ut-assert does not provide its full context. It only provides the first argument, EventId, providing no way to for unit tests to verify the correct EventType or event message text was sent.

**To Reproduce**
Steps to reproduce the behavior:
1. See the [CFE_EVS_SendEvent stub](https://github.com/nasa/cFE/blob/60a5f6580d1e737f46efce0418aba5b8dac6df6a/fsw/cfe-core/ut-stubs/ut_evs_stubs.c#L106). Only EventId is registered with the context and copied. 

**Expected behavior**
The CFE_EVS_SendEvent stub should provide its full context for unit testing purposes. My preference would be that the event message text provided would not include the format specifiers and instead be the resulting string.

**Code snips**
```
int32 CFE_EVS_SendEvent(uint16 EventID,
                        uint16 EventType,
                        const char *Spec,
                        ...)
{
    int32 status;

    UT_Stub_RegisterContext(UT_KEY(CFE_EVS_SendEvent), &EventID);
    status = UT_DEFAULT_IMPL(CFE_EVS_SendEvent);

    if (status >= 0)
    {
        UT_Stub_CopyFromLocal(UT_KEY(CFE_EVS_SendEvent), (uint8*)&EventID, sizeof(EventID));
    }

    return status;
}
```

**System observed on:**
 - cFE 6.7.11

**Additional context**
n/a

**Reporter Info**
Eric Gilligan NASA/GSFC-5820

gilliganeric, 2020-04-10:
Thinking about this more, I'd also be fine with the Context getting the string as passed into `CFE_EVS_SendEvent` with the format specifiers still in it, as long as the extra arguments are put into the Context as well.

skliper, 2020-04-10:
I think @jphickey has examples for getting format and arguments.

jphickey, 2020-04-10:
Yes, this stub will need an update to pass through more context if UT test cases need this.

The framework has the facility to do this, via the `UT_DefaultStubImplWithArgs` routine, which can pass through the full context in a va_list to a hook implementation.   Then in the hook implementation one can do whatever is needed.

I'd still recommend against passing it to [v]snprintf, as that is now crossing the bridge of no longer validating the direct outputs of the unit, but rather mimicking/duplicating something that the FSW might do and validating that.  Very different thing.  (stubs shouldn't duplicate FSW logic).

gilliganeric, 2020-04-10:
I agree. Disregard that preference in the original description. 

Would this look like `ArgPtr`'s last element in the hook function's `UT_StubContext_t` being a `va_list` containing the variable number of arguments?

AlanSGibson, 2020-04-10:
I completely agree with not passing it to [v]snprintf, but I would like to see the default implementation of the CFE_EVS_SendEvent stub make the arguments available for checking without having to use a hook.

jphickey, 2020-04-10:
The "Fixed" arguments, up to and including the spec string, can be registered using calls to `UT_Stub_RegisterContext()`.

The use `va_start` to capture all the remaining variable arguments, and invoke `UT_DefaultStubImplWithArgs` with va list as the final argument, then do `va_end` after it returns.

For an example of a stub that uses this mechanism, see `OS_printf` in: https://github.com/nasa/osal/blob/master/src/ut-stubs/osapi-utstub-printf.c

Usage with hook function example is in sample_lib: https://github.com/nasa/sample_lib/blob/master/unit-test/coveragetest/coveragetest_sample_lib.c

skliper, 2020-04-15:
20200415 - Ok to add and pass it through

skliper, 2020-05-12:
@gilliganeric - can you confirm #673 does what you need, or clarify what would still be necessary?

ericgilligan-nasa, 2020-05-14:
It looks like #673 provides the Event Type and Spec, but not the variable arguments as a va_list as described by @jphickey earlier. Is that correct?

skliper, 2020-05-14:
Correct, we can leave this open until that's implemented:

> The use `va_start` to capture all the remaining variable arguments, and invoke `UT_DefaultStubImplWithArgs` with va list as the final argument, then do `va_end` after it returns.
>
> For an example of [...]

jphickey, 2020-05-14:
Ahh yes, sorry I overlooked that during my review of the change.  If you want to have the actual data available in your hook this needs to use the `UT_DefaultStubImplWithArgs()` function to pass them all through.

PR #713 (2020-06-02): Fix #603, Full context info for EVS event stubs
**Describe the contribution**
Create a va_list object to capture the full set of variable arguments passed to CFE_EVS_SendEvent and variants (TimedEvent, WithAppID).  Use the variable argument form of the stub implementation to pass all info through to a (possible) hook function.

Fixes #603 

**Testing performed**
Build all unit tests and confirm normal operation, all passing.

**Expected behavior changes**
Hook functions may now use the va_list form and obtain the full context info.

**System(s) tested on**
Ubuntu 20.04

**Additional context**
 [... cut]
commit 0e3a00a4 (2020-06-02): Merge pull request #713 from jphickey/fix-603-add-evs-stub-args

Fix #603, Full context info for EVS event stubs
commit 234da0af (2020-06-02): Fix #603, add va_list for full stub context

Add a va_list to capture the full set of variable arguments
on all CFE_EVS event calls.

Also add va_list for CFE_ES_WriteToSyslog
PR #729 (2020-06-10): Integration Candidate: 2020-05-27
**Describe the contribution**
Fix #704
Fix #603
Fix #719

**Testing performed**
https://github.com/nasa/cFS/pull/96/checks

**Expected behavior changes**
PR #714 -  API Change: cFE ut_sb_stubs now has CFE_SB_DeletePipe available.
    Behavior Change: App unit tests requiring this will not fail to build due to undefined reference to `CFE_SB_DeletePipe`

PR #713 -  Hook functions may now use the `va_list` form and obtain the full set of variable arguments passed to `CFE_EVS_SendEvent` and variants. 

PR #725 - Replace all direct references to data types defined  [... cut]
closed 2020-06-10 by astrogeco