nasa/cFE #608: CFE_SB_GetLastSenderID returns pointer to internal data, possible race

skliper, 2020-04-14:
**Describe the bug**
CFE_SB_GetLastSenderID returns pointer to data that could be overwritten at any time

See https://github.com/nasa/cFE/pull/605#discussion_r407649126 for related comments

**To Reproduce**
Conceptually - app calls CFE_SB_GetLastSenderID, gets pointer, data gets overwritten, app takes action based on overwritten data vs original context

**Expected behavior**
No race.

**Code snips**
https://github.com/nasa/cFE/blob/60a5f6580d1e737f46efce0418aba5b8dac6df6a/fsw/cfe-core/src/sb/cfe_sb_api.c#L1651-L1657

**System observed on:**
N/A

**Additional context**
#605

**Reporter Info**
Jacob Hageman

skliper, 2020-04-14:
Ping @CDKnightNASA @tngo67 and any other user's of this API (SBN related?)

astrogeco, 2020-08-19:
See https://github.com/nasa/cFE/pull/816

PR #784 (2020-08-14): fix #759 - deprecates GetLastSenderId()
**Describe the contribution**
Fixes #759 - deprecates CFE_SB_GetLastSenderId() API.
Fixes #608

**Testing performed**
make with and without OMIT_DEPRECATED defined.

**System(s) tested on**
Debian 10.3

**Contributor Info - All information REQUIRED for consideration of pull request**
Christopher.D.Knight@nasa.gov
closed 2020-08-19 by astrogeco