nasa/cFE #618: Unit tests fail to build when MESSAGE_FORMAT_IS_CCSDS_VER_2 is enabled

johnphamngc, 2020-04-15:
**Describe the bug**
Unit tests fail to build when MESSAGE_FORMAT_IS_CCSDS_VER_2 is enabled

**To Reproduce**
Enable MESSAGE_FORMAT_IS_CCSDS_VER_2 in `*mission_cfg.h`
Run `make CFLAGS="-m32 -Wno-format" SIMULATION=native ENABLE_UNIT_TESTS=true` 

**Expected behavior**
Compilation succeeds, however build actually fails due to missing `#include`

**Code snips**
This can be fixed by doing the following 
```
--- a/fsw/cfe-core/ut-stubs/ut_sb_stubs.c
+++ b/fsw/cfe-core/ut-stubs/ut_sb_stubs.c
@@ -34,6 +34,8 @@
 */
 #include <string.h>
 #include "cfe.h"
+#include "cfe_platform_cfg.h"
+#include "../sb/cfe_sb_msg_id_util.h"
 #include "utstubs.h"
```
However, subsequently a checksum error is encountered when running the tests.
This was worked around by doing the following, but unsure if the workaround is correct.
```
--- a/fsw/cfe-core/unit-test/sb_UT.c
+++ b/fsw/cfe-core/unit-test/sb_UT.c
@@ -9456,7 +9456,7 @@ void Test_CFE_SB_ChecksumUtils(void)
 #ifndef MESSAGE_FORMAT_IS_CCSDS_VER_2
     ExpRtnFrmGet = 0x2f;
 #else
-    ExpRtnFrmGet = 0x65;
+    ExpRtnFrmGet = 0x61;
 #endif

     /* Validation expected to return true */
```

**System observed on:**
 - Hardware: N/A
 - OS: RHEL7
 - Versions cFE 6.7.12

**Additional context**
Add any other context about the problem here.

**Reporter Info**
John N Pham, Northrop Grumman

johnphamngc, 2020-04-22:
It looks like Test_CFE_SB_ChecksumUtils has further issues w/ CCSDSv2, as it assumes SID = MsgID in multiple cases, and doesn't account for the fact that ApidQSystemId is populated w/ the spacecraft ID by CFE_SB_InitMsg (which is a user configurable value, causing ExpRtnFrmGet to be wrong if the spacecraft ID is changed from the defaults).

jphickey, 2020-04-22:
I mistakenly filed a duplicate ticket about this because I just discovered it myself.  Looks like there are a few issues in here.

PR #674 (2020-05-08): Fix #618, stubs must not depend on real msgid implementation
**Describe the contribution**
This makes the SB stubs which access message structures into actual stubs, not a replica of the normal implementation.  Stubs manipulate a local (stored in the UT framework) out-of-band buffer to hold the metadata about the message.

This removes the dependency on the actual definition of MsgId used by the mission and makes them agnostic to the setting of extended headers.

This revealed a few other minor issues in test cases where they were depending on values sitting in globals (also fixed).

Fixes #61 [... cut]
commit 8642f8ac (2020-05-08): Fix #618, stubs must not depend on real msgid implementation

The CFE_SB_GetMsgId/SetMsgId functions, among others, were implemented
as replicas of the actual FSW code.  This creates a dependency on the
actual definition of MsgId used by the mission.

This makes the stub and actual stub.  Stubs not actually
read/write to the message in any way, they just manipulate a
local (stored in the UT framework) out-of-band buffer to hold
the metadata about the message.

This revealed a few other minor issues in test cases where they
were depending on values sitting in globals (fixed).
commit d2c54cc0 (2020-05-08): Merge pull request #674 from jphickey/fix-618-stub-msgid-dependency

Fix #618, stubs must not depend on real msgid implementation
PR #692 (2020-05-13): Integration Candidate COMBINED 2020-04-29 and 2020-05-06
**Describe the contribution**

Fix #399, Deprecate decompress
Fix #411, rework exception handling in CFE
Fix #484, Deprecate output to shell commandFix #491, Add UT macros
Fix #519, Deprecate CFS/FS time conversion APIs
Fix #523, SB Subscription report control on separate MID
Fix #551, Remove duplicate CFE_TIME_Local1HzISR prototype
Fix #559, Resolve doxygen warnings
Fix #618, stubs must not depend on real msgid implementation
Fix #637, Remove old CFE_ARINC653-related tests
Fix #639, Remove ES_APP_DEBUG functionality
Fix #641, string ope [... cut]
closed 2020-05-13 by astrogeco