nasa/cFE #620: Question on Table Service update procedure and notifications by message

pcooksey, 2020-04-16:
**Is your feature request related to a problem? Please describe.**
I guess some more documentation on this procedure would be beneficial when talking about the table services. This is related to issue #130 but more on documenting that feature. Specifically, there really isn't a lot of documentation on notify by message function:

> int32 CFE_TBL_NotifyByMessage(CFE_TBL_Handle_t TblHandle, CFE_SB_MsgId_t MsgId, uint16 CommandCode, uint32 Parameter);

We want to use this to not have to poll for changes which works but I am slightly confused on the notification process.

**Describe the solution you'd like**
I have enabled notify by message and I execute the following to test the app:
1. Upload new file to "spacecraft" (Just a new table with the table's exact name in it)
2. Send command to load table (this goes into inactive buffer):
   1. ./cmdUtil --endian=LE --host=localhost --port=1234 --pktid=0x1804 --cmdcode=2 --string="064:/cf/apps/SampleAppletTb2.tbl"
3. Send command to validate table in inactive buffer
   1. ./cmdUtil --endian=LE --host=localhost --port=1234 --pktid=0x1804 --cmdcode=4 --half=0 --string="040:SampleApplet_App.SATable1" 
   2. **Note: this is the first time our App will be notified something has changed. Yet it can't do anything because the table is in the inactive buffer (the app sees there is no new info). This seems like a strange notification to receive. Why?**
4. Send command to active table
    1. ./cmdUtil --endian=LE --host=localhost --port=1234 --pktid=0x1804 --cmdcode=5 --string="040:SampleApplet_App.SATable1"
    2. **Note: this will also notify our App and this time our App will be given the signal that there is information that needs to be updated.**
	
Questions:
First, is this the ideal process for updating tables in flight? I mostly had to piece together this by looking through the public headers. If I happened to just mistakenly miss some documentation on this then a pointer to that documentation would be awesome! 

Second, why is there a message after the validation function if the app can't do anything since the new table is still in the inactive buffer. Is this a way to potentially ask the app to release the table pointer if it had been holding on to it? 

**Describe alternatives you've considered**
N/A

**Additional context**
Maybe there should be a issue format for questions? 

**Requester Info**
Philip Cooksey, NASA Ames

Thank you for your time and help!

pcooksey, 2020-04-16:
It seems `CFE_TBL_Manage` was obscuring the fact that the validation of the table was happening by our App since I was getting CFE_SUCCESS I didn't think anything happened. But if I ask for the status before using the manage function then I can see that it is `CFE_TBL_INFO_VALIDATION_PENDING`. This makes more sense now. Is this correct?

skliper, 2020-04-22:
Yes, pending until managed.

closed 2020-04-22 by skliper