nasa/cFE #621: Memory alignment issue in ES

skliper, 2020-04-16:
**Is your feature request related to a problem? Please describe.**
```
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/es/cfe_es_shell.c: In function 'CFE_ES_ShellOutputCommand':
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:178: error: cast increases required alignment of target type
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:179: error: cast increases required alignment of target type
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:208: error: cast increases required alignment of target type
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:209: error: cast increases required alignment of target type
```

**Describe the solution you'd like**
Resolve bug.

**Describe alternatives you've considered**
None.

**Additional context**
Other shell issues, #484, #84, etc.

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-04-16:
I likely have something related to this in my stash of unmerged fixes, because I had something for #313 and #314 too.

skliper, 2020-04-16:
Want to submit something for those three?  Needed ASAP (blocking cert work).

jphickey, 2020-04-16:
Right now focusing on nasa/osal#285 stuff, but if it can wait until next week, then I can submit something.

jphickey, 2020-04-16:
Note that `-Wcast-align` was something we added recently, it never compiled cleanly on Vxworks/PPC.  Whenever testing for vxworks I simply turned this warning off in the local build.

closed 2020-05-08 by astrogeco