nasa/cFE #626: Utilize CMake interface libraries

skliper, 2020-04-20:
**Is your feature request related to a problem? Please describe.**
See conversation around https://github.com/nasa/cFE/pull/585#discussion_r411528313

Interface libraries may be a cleaner approach than current implementation

**Describe the solution you'd like**
Consider utilizing interface libraries

**Describe alternatives you've considered**
Leave as-is

**Additional context**
See #585 

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-11-02:
If we do this, should add cmake_minimum_required(VERSION 3.0.0) so people using CentOS 7 which comes with CMake 2.8 won't get the confusing error message.

jphickey, 2020-11-02:
Unfortunately RHEL 7.x is still the "standard" baseline - many enterprises deploy this version and don't give users much choice in the matter, because it is viewed as "stable" - even though the likely explanation is that its components are so old nothing works with it.

Until RHEL/CentOS 8.x gets more adoption we probably shouldn't do anything which requires CMake 3.x.  It looks like RHEL 7.7 is EOL in August 2021 which is apparently the last planned release in the 7.x series.

Unless there is a fairly easy/straightforward way to update CMake via the EPEL repos?

skliper, 2020-11-02:
> Unless there is a fairly easy/straightforward way to update CMake via the EPEL repos?

cmake3 is available to install.

jphickey, 2020-11-02:
> cmake3 is available to install.

If there is a CMake 3 package available then sure - we should require that version!  (and ideally also point users to instructions to install it, as I know the current/default install steps still yield version 2.8.12)

My vote would be to set `cmake_minimum_required` to whatever version is shipped with the oldest still-active/non-EOL RHEL/CentOS version - this distro is usually a good reference point for the most obsolete versions of any particular tool.