nasa/cFE #629: make prep broken out of the box

skliper, 2020-04-21:
**Describe the bug**
Prep fails out of the box when following README instructions:

```
-- Configuring for system arch: cpu1
-- The C compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- OMIT_DEPRECATED=false: Deprecated elements included in build
Error: "" is not a valid BSP type
CMake Error at /export/home/jhageman/cFS/cFS-GitHub/osal/CMakeLists.txt:66 (message):
  OSAL_SYSTEM_BSPTYPE must be set to the appropriate BSP
```

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the README instructions, make prep fails.

**Expected behavior**
Expected it to default to pc-linux as in the past.

**Code snips**
The following does not set OSAL_SYSTEM_BSPTYPE since CFE_SYSTEM_PSPNAME and OSAL_SYSTEM_OSTYPE is defined...

https://github.com/nasa/cFE/blob/0648a47bf4369138c31f9019458ab2eceec738a8/cmake/arch_build.cmake#L275-L289

**System observed on:**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: master bundle

**Additional context**
Short term work around - pass in SIMULATION=native at prep, or likely can define the target system in targets.cmake (2nd approach not tested yet)

**Reporter Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-04-21:
Looks like the sample toolchains set OSAL_SYSTEM_BSPNAME, not OSAL_SYSTEM_BSPTYPE.  I don't really follow the logic for the "guess" in arch_build.cmake.  Seems like it would be safer just to error if any of the three aren't defined, otherwise it may overwrite customization set in the toolchain file (require all 3 to be defined, don't overwrite from deep in the build if 1 is missing).

stanislaw, 2020-04-21:
I had this today too:

Was anything renamed recently?

```diff
--- a/sample_defs/toolchain-cpu1.cmake
+++ b/sample_defs/toolchain-cpu1.cmake
@@ -20,6 +20,6 @@ SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE   NEVER)
 # These variable settings are specific to cFE/OSAL and determines which
 # abstraction layers are built when using this toolchain
 SET(CFE_SYSTEM_PSPNAME      "pc-linux")
-SET(OSAL_SYSTEM_BSPNAME     "pc-linux")
+SET(OSAL_SYSTEM_BSPTYPE     "pc-linux")
 SET(OSAL_SYSTEM_OSTYPE      "posix")
```

jphickey, 2020-04-21:
The difference is that in the current master branch `OSAL_SYSTEM_BSPTYPE` is required to be specified, whereas before it was ignored and only `CFE_SYSTEM_PSPNAME` was used.  Now the PSP and OSAL BSP are used together so they are both required.  The mismatch/typo in the example toolchain-cpuX.cmake has existed for a long time but it never mattered until now

PR #634 (2020-04-24): Fix #632, Infer OSAL_SYSTEM_BSPTYPE from CFE_SYSTEM_PSP_NAME
**Describe the contribution**
Normally the CFE PSP uses/links with an OSAL BSP of the same name. This removes the need to explicitly specify OSAL_SYSTEM_BSPTYPE in the toolchain file, as it can be reliably inferred.

Fixes #632 

This is also another alternative way to fix #629 

**Testing performed**
Build with defaults (as in README, e.g. a simple `make prep`), as well as `SIMULATION=native`, and for i686-rtems4.11 platform.  Confirm successful build.

Also "forced" a mismatch by hacking the i686-rtems4.11 toolchain to request bad  [... cut]
closed 2020-05-19 by skliper