nasa/cFE #631: split unit-test files

CDKnightNASA, 2020-04-21:
**Is your feature request related to a problem? Please describe.**
Using the Eclipse IDE, the large size of the unit test source files triggers the "Scalability mode".

**Describe the solution you'd like**
There's little reason the source files need to be monolithic, splitting the unit test code (further) into separate files organized by API function or groups of related functions will improve organization, readability and performance.

**Describe alternatives you've considered**
Leaving the files as-is.

**Requester Info**
Christopher.D.Knight@nasa.gov

jphickey, 2020-04-21:
My recommendation is to follow the pattern in OSAL and make a 1:1 relationship between UT files and FSW source units.  Each UT unit should strictly cover the functions in the corresponding FSW unit, nothing else.

This actually makes linking a lot easier....

pavll, 2021-07-25:
Hello @CDKnightNASA @jphickey 

I'm new to NASA's OpenSource projects, but I'm very interested in the software that runs on satellites, rovers, and other NASA objects in space. Which is why I would like to contribute my part to this repo. I found this issue under "good first issue" and wanted to ask if it is still up to date?

If so, could you please help me get started by telling me exactly what unit tests you are talking about here?
I understood that you would like to split the UTs from "fs/ut-coverage/fs_UT" into different files to match those in "fs/fsw/src/**".

Translated with www.DeepL.com/Translator (free version)