nasa/cFE #636: Unit tests fail to build when extended headers enabled

jphickey, 2020-04-22:
**Describe the bug**
As a test of the most recent integration candidate, I attempted to verify operation with extended headers but the unit test code fails to build in this configuration.

The following errors exist in `ut_sb_stubs.c`:
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c: In function ‘CFE_SB_GetMsgId’:
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c:293:31: error: ‘CFE_SB_CMD_MESSAGE_TYPE’ undeclared (first use in this function); did you mean ‘CFE_SB_NO_MESSAGE’?
               MsgId = MsgId | CFE_SB_CMD_MESSAGE_TYPE;
                               ^~~~~~~~~~~~~~~~~~~~~~~
                               CFE_SB_NO_MESSAGE
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c:293:31: note: each undeclared identifier is reported only once for each function it appears in
In file included from /home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/cfe_sb.h:43:0,
                 from /home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/cfe_evs_msg.h:41,
                 from /home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/cfe_evs.h:42,
                 from /home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/cfe.h:51,
                 from /home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c:36:
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c: In function ‘CFE_SB_SetMsgId’:
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c:500:42: error: implicit declaration of function ‘CFE_SB_RD_APID_FROM_MSGID’; did you mean ‘CFE_SB_INVALID_MSG_ID’? [-Werror=implicit-function-declaration]
   CCSDS_WR_APID(MsgPtr->SpacePacket.Hdr, CFE_SB_RD_APID_FROM_MSGID(MsgId) );
                                          ^
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/ccsds.h:294:92: note: in definition of macro ‘CCSDS_WR_APID’
 #define CCSDS_WR_APID(phdr,value)  ((((phdr).StreamId[0] = ((phdr).StreamId[0] & 0xF8) | ((value >> 8) & 0x07))) ,\
                                                                                            ^~~~~
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c:502:42: error: implicit declaration of function ‘CFE_SB_RD_TYPE_FROM_MSGID’; did you mean ‘CFE_SB_HK_TLM_MSG’? [-Werror=implicit-function-declaration]
   CCSDS_WR_TYPE(MsgPtr->SpacePacket.Hdr, CFE_SB_RD_TYPE_FROM_MSGID(MsgId) );
                                          ^
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/ccsds.h:305:91: note: in definition of macro ‘CCSDS_WR_TYPE’
 #define CCSDS_WR_TYPE(phdr,value)   ((phdr).StreamId[0] = ((phdr).StreamId[0] & 0xEF) | ((value << 4) & 0x10))
                                                                                           ^~~~~
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c:509:46: error: ‘CFE_PLATFORM_ENDIAN’ undeclared (first use in this function)
   CCSDS_WR_ENDIAN(MsgPtr->SpacePacket.ApidQ, CFE_PLATFORM_ENDIAN);
                                              ^
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/ccsds.h:350:110: note: in definition of macro ‘CCSDS_WR_ENDIAN’
 #define CCSDS_WR_ENDIAN(shdr,val)        ( (shdr).APIDQSubsystem[0] = ((shdr).APIDQSubsystem[0] & 0xFB) | (((val) & 0x01) << 2) )
                                                                                                              ^~~
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/ut-stubs/ut_sb_stubs.c:513:52: error: implicit declaration of function ‘CFE_SB_RD_SUBSYS_ID_FROM_MSGID’; did you mean ‘CFE_SB_ALLSUBS_TLM_MSG’? [-Werror=implicit-function-declaration]
   CCSDS_WR_SUBSYSTEM_ID(MsgPtr->SpacePacket.ApidQ, CFE_SB_RD_SUBSYS_ID_FROM_MSGID(MsgId));
                                                    ^
/home/joe/code/cfecfs/github/cfe/fsw/cfe-core/src/inc/ccsds.h:353:109: note: in definition of macro ‘CCSDS_WR_SUBSYSTEM_ID’
 #define CCSDS_WR_SUBSYSTEM_ID(shdr,val)  (((shdr).APIDQSubsystem[0] = ((shdr).APIDQSubsystem[0] & 0xFE) | ((val & 0x0100) >> 8)) ,\
                                                                                                             ^~~
**To Reproduce**
Build according to instructions, but set:
`#define MESSAGE_FORMAT_IS_CCSDS_VER_2` 
in mission config.

**Expected behavior**
Build should succeed.

**System observed on:**
Ubuntu 18.04 LTS 64-bit

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-04-22:
This seems to stem from the fact that the `CFE_SB_SetMsgId` "stub" is actually (more or less) a copy of the real implementation which doesn't understand the macros used in the MESSAGE_FORMAT_IS_CCSDS_VER_2 path.

Stubs shouldn't be real implementations, they are stubs.  Should try to make these the same regardless of CCSDS version.

jphickey, 2020-04-22:
Just realized this is a duplicate of #618, apologies...

closed 2020-04-22 by jphickey

skliper, 2020-04-23:
> This seems to stem from the fact that the `CFE_SB_SetMsgId` "stub" is actually (more or less) a copy of the real implementation which doesn't understand the macros used in the MESSAGE_FORMAT_IS_CCSDS_VER_2 path.
> 
> Stubs shouldn't be real implementations, they are stubs. Should try to make these the same regardless of `MESSAGE_FORMAT_IS_CCSDS_VER_2` setting.

I'd like to deprecate all the "real implementation" aspects in the stubs.