nasa/cFE #643: Add "special" ES shell commands as real commands

skliper, 2020-04-27:
**Is your feature request related to a problem? Please describe.**
ES special commands using the SHELL FC and a special string don't follow command processing model, missing requirements, etc
CFE_ES_ListApplications
CFE_ES_ListTasks
CFE_ES_ListResources

**Describe the solution you'd like**
Implement these as real commands

**Describe alternatives you've considered**
Delete these

**Additional context**
#484 

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-04-27:
Other options - Could add an event as part of QueryAllCmd (prints name to replace CFE_ES_ListApplications), QuerryAllTasksCmd (prints task name/id, app name/id to replace CFE_ES_ListTasks).  

Note CFE_ES_ListResources still seems unique/useful but it only does counts.  Might make more sense to include names or possibly other info... or maybe would make even more sense to include resource counts in the housekeeping message?  It's got counts for core apps, external apps, tasks, libs, why not resources?  Or maybe just a by request packet (add new) that has all the counts?  Send to a file doesn't really seem to have that much of a benefit.

astrogeco, 2020-04-29:
CCB 2020-04-29 : Discussed

skliper, 2020-04-29:
Note - no strong opinions expressed, seems like approach in the comments above would be beneficial to users.  Suggesting that as a path forward (events for the name/tasks as part of the already existing commands, new more informative output for resources).

igorluppi, 2024-05-21:
Is this thread still open for draco-rc4?