nasa/cFE #644: Global build options no longer global

skliper, 2020-04-27:
**Describe the bug**
Deprecation flags not showing up in build for cFE/apps/PSP

**To Reproduce**
Prep with SIMULATION=native ENABLE_UNIT_TEST=true OMIT_DEPRECATED=true
make VERBOSE=true and see the DEPRECATED flags not applied to cFE/apps/PSP

**Expected behavior**
Global flags should be global

**Code snips**
None

**System observed on:**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: Bundle (was testing fix to #484)

**Additional context**
Critical for CI testing...

**Reporter Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-04-27:
Diagnosis is that the recent change for #585 (flag propagation) needs to append rather than set.  

Fix is pushed directly to "integration-candidate" branch (fairly trivial).  See commit above for review.

skliper, 2020-04-27:
> Diagnosis is that the recent change for #585 (flag propagation) needs to append rather than set.
> 
> Fix is pushed directly to "integration-candidate" branch (fairly trivial). See commit above for review.

Thanks for the quick fix.  Worked for me.  Linked issue directly to IC PR.

commit b1ef9fc6 (2020-04-27): Fix #644, Append directory properties

The set_directory_properties() function overwrites existing properties.
Normally the properties were empty but the global_build_options.cmake
might make it non-empty (e.g. OMIT_DEPRECATED).

Using set_property() signature instead has an option to APPEND rather
than overwrite.
closed 2020-04-27 by astrogeco