nasa/cFE #651: CFE_ES_SEND_MEM_POOL_STATS_CC Issues

dmknutsen, 2020-04-29:
**Describe the bug**
There are multiple issues with the CFE_ES_SEND_MEM_POOL_STATS_CC command:
1.	The command/function(s) used are fundamentally flawed in that a user can send a seemingly benign command that can result in a segmentation fault if one of the command parameters is incorrect.  For example, if the PoolHandle parameter is set to zero a segmentation fault will result (pretty much any value below xFFFFFE28 faults on my machine).  This occurs when handle is validated via the CFE_ES_ValidateHandle function.  Should consider modifying the CFE_ES_ValidateHandle function + updating the unit test to test command on boundary/extreme conditions.
2.	On a 64-bit machine if the PoolHandle parameter is set to a valid value the function will fail.  This is because CFE_PSP_MemValidateRange – which is called via the CFE_ES_ValidateHandle function as part of the validation process, limits the max memory range of the handle to xFFFFFFFE.
3.	On a 64-bit machine - compiler added padding will be applied to the command.  The order of variable declaration should ideally be descending in size to avoid future conflicts.

**Expected behavior**
Command works nominally and is vetted via combination of unit/functional tests.

**System observed on:**
Oracle VM VirtualBox
OS: ubuntu-19.10
Versions: cFE 6.7.13.0, OSAL 5.0.12.0, PSP 1.4.9.0

**Reporter Info**
Dan Knutsen
NASA/Goddard

skliper, 2020-08-05:
@jphickey want to take this one?

jphickey, 2020-08-05:
Sure, sounds good to me.  My proposal (if others agree) would be to apply the same paradigm I'm using for CFE_ES_AppID -- define a type (NOT an address) which abstractly identifies a memory pool, e.g. `CFE_ES_PoolID_t` ...  then this gets put into CMD/TLM packets to identify the memory pool.  The CFE_ES then looks up this value to get the REAL memory address of the pool.

jphickey, 2020-08-05:
Actually, since we already have a `CFE_ES_MemHandle_t` type -- then no new type is needed -- can just use this, but make it an abstract ID rather than a memory address.

skliper, 2020-08-06:
Perfect.  Hopefully we don't have any other memory addresses in commands...

PR #917 (2020-10-01): Fix #651, Use resource ID for memory pools
**Describe the contribution**

Instead of identifying a memory pool by its memory address, use a resource ID.  IDs are a constant size, regardless of
whether the host machine is 32 or 64 bits.

- IDs can be put into commands/telemetry and maintain a more consistent format with consistent alignment requirements.
- IDs can be independently verified without dereferencing memory.  Previously the only way to validate a memory pool was to read the address pointed to, which results in a segfault if the address was bad.

Fixes #651 

**Testing performed**
Bui [... cut]
commit 3a809eb4 (2020-10-01): Merge pull request #917 from jphickey/fix-651-abstract-mempool-id

Fix #651, Use resource ID for memory pools
commit f6ae2fbe (2020-10-01): Fix #651, Use resource ID for memory pools

Instead of identifying a memory pool by its memory address,
use a resource ID.  IDs are a constant size, regardless of
whether the host machine is 32 or 64 bits.

- IDs can be put into commands/telemetry and maintain a more
  consistent format with consistent alignment requirements.
- IDs can be independently verified without dereferencing
  memory.  Previously the only way to validate a memory pool
  was to read the address pointed to, which results in a SEGV
  if the address was bad.
PR #936 (2020-10-06): Integration Candidate 2020-10-07
**Describe the contribution**
Fix #923 
Fix #897  
Fix #651
Fix #923
Fix #705
Fix #888

**Testing performed**
See Bundle: https://github.com/nasa/cFS/pull/145/checks

**Expected behavior changes**

PR #898 - Adds the field `UnregAppID` to track whether an "unregistered" event was generated, un-overloading the `EventCount` field to serve its primary purpose of counting actual events generated from a valid/registered AppID.

PR #925 - Move the AppID lookup execution to be early in the CFE_SB_SendMsgFull implementation. This avoids double locking between SB+ES and [... cut]
closed 2020-10-06 by astrogeco