nasa/cFE #652: CFE needs to provide CMake-based config file for OSAL

jphickey, 2020-04-29:
**Is your feature request related to a problem? Please describe.**
Currently the CFE cmake script generates an `osconfig.h` file for OSAL to compile with.

**Describe the solution you'd like**
After nasa/osal#285 is fixed (PR nasa/osal#427) the CFE will need to  pass OSAL a configuration file in CMake syntax, which then OSAL uses to generate its own `osconfig.h`.

**Additional context**
Needed for compatibility

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

closed 2020-05-05 by astrogeco

jtovias, 2020-05-20:
@jphickey , I'm trying to override OS_MAX_FILE_NAME, how do I update OS_MAX_FILE_NAME in native_osconfig.cmake?

jtovias, 2020-05-20:
@jphickey figured it out. have to set OSAL_CONFIG_MAX_FILE_NAME.

PR #654 (2020-04-30): Fix #652, cmake-based OSAL config
**Describe the contribution**
Rather than CFE generating an osconfig.h file directly, it just needs to select file(s) from the defs directory and pass them to OSAL for the config.

This makes it compatible with the upcoming nasa/osal#285 change (in PR nasa/osal#427)

Fixes #652

**Testing performed**
Build code with default provided and confirm that OSAL generates the correct config (osconfig.h) and the build succeeds

**Expected behavior changes**
Users must now select OSAL options via the CMake file in their defs directory, rather than the `osconfig.h` file. [... cut]
commit 8cafba7e (2020-04-30): Fix #652, cmake-based OSAL config

Rather than CFE generating an osconfig.h file directly, it just
needs to select file(s) from the defs directory and pass them
to OSAL for the config.
commit ec38f091 (2020-04-30): Merge pull request #654 from jphickey/fix-652-osal-config

Fix #652, cmake-based OSAL config
PR #667 (2020-05-04): Fix #652, cmake-based OSAL config
**Describe the contribution**
Rather than CFE generating an osconfig.h file directly, it just needs to select file(s) from the defs directory and pass them to OSAL for the config.

This makes it compatible with the upcoming nasa/osal#285 change (in PR nasa/osal#427)

Fixes #652

**Testing performed**
Build code with default provided and confirm that OSAL generates the correct config (osconfig.h) and the build succeeds

**Expected behavior changes**
Users must now select OSAL options via the CMake file in their defs directory, rather than the `osconfig.h` file. [... cut]