nasa/cFE #657: Missing registering contexts in stub functions

jtovias, 2020-04-30:
The issue causes seg faults cause by not registering contexts. I resolved this by adding the missing registrations.
 

cfe/fsw/cfe-core/ut-stubs/ut_evs_stubs.c:
int32 CFE_EVS_SendEvent(uint16 EventID,
                        uint16 EventType,
                        const char *Spec,
                        ...)
{
    int32 status;
    UT_Stub_RegisterContext(UT_KEY(CFE_EVS_SendEvent), &EventID);
    UT_Stub_RegisterContext(UT_KEY(CFE_EVS_SendEvent), &EventType);  <----- missing
    UT_Stub_RegisterContext(UT_KEY(CFE_EVS_SendEvent), Spec);              <----- missing
..
}
 
cfe/fsw/cfe-core/ut-stubs/ ut_es_stubs.c:
int32 CFE_ES_WriteToSysLog(const char *pSpecString, ...)
{
    int32   status;
    UT_Stub_RegisterContext(UT_KEY(CFE_ES_WriteToSysLog), pSpecString);  <----- missing
 ...}

jtovias, 2020-05-19:
@skliper , Do I make a branch from development and call it fix657-add-UT-Stub_RegisterContext?

skliper, 2020-05-19:
@jtovias - I was just marking that you already fixed it in #673, marking is more for internal documentation/compliance and VDD generation at release time

PR #673 (2020-05-20): Fix #657, Add UT_Stub_RegisterContext to CFE_EVS_SendEvent and CFE_ES_WriteToSysLog
Add missing UT_Stub_RegisterContext to CFE_EVS_SendEvent and CFE_ES_WriteToSysLog.
This will create the correct register buffer size.

**Describe the contribution**
Add missing UT_Stub_RegisterContext to CFE_EVS_SendEvent and CFE_ES_WriteToSysLog.
This will create the correct register buffer size.
- Fixes #657

**Testing performed**
Steps taken to test the contribution:

travis CI passed:    https://travis-ci.com/github/jtovias/cFS/builds/164052416

**Expected behavior changes**
A clear and concise description  [... cut]
commit 57277815 (2020-05-20): Merge pull request #673 from jtovias/fix657-add-UT-Stub-RegisterContext

Fix #657, Add UT_Stub_RegisterContext to CFE_EVS_SendEvent and CFE_ES_WriteToSysLog
commit 84e10c58 (2020-05-20): Fix #657, Register context in SendEvent and SysLog stubs

Add missing UT_Stub_RegisterContext to CFE_EVS_SendEvent and CFE_ES_WriteToSysLog.
This will create the correct register buffer size.
PR #712 (2020-05-22): Integration Candidate 2020-05-13
**Describe the contribution**
Fix #309 
Fix #657 
Fix #679
Fix #681  
Fix #684  
Fix #707

**Testing performed**
Bundle CI - https://travis-ci.com/github/nasa/cFS/builds/167639047

**Expected behavior changes**
PR #673 - using ut stubs CFE_EVS_SendEvent and CFE_ES_WriteToSysLog, the register buffer will have the correct size. access to register buffer element will exist

PR #685 - Both the main task and the child task(s) are successfully deleted and restarted after the exception occurs.

PR #686 -  Fixes doxygen warnings for the tbl subsystem.

PR #691 - No com [... cut]
closed 2020-05-22 by astrogeco