nasa/cFE #659: CFE_SB_GetMsgId returns 0.

jtovias, 2020-04-30:
**Describe the bug** 
CFE_SB_GetMsgId in cfe/fsw/cfe-core/ut-stubs/ ut_sb_stubs.c.
 
CFE_SB_GetMsgId is not returning the PLSS_STATE_DET_WAKEUP_MID set by UT_SetForceFail. The problem I see is the MsgId is now calculated by the MsgPtr which points to a Buffer.Msg (defined in CFE_SB_RcvMsg) but this Buffer.Msg (StreamId[0] and streamId[1]) is set to all zeros and never set to the PLSS_STATE_DET_WAKEUP_MID. In cfe6.6, the MsgId was calculated from the StubEntry data which is set by  UT_SetForceFail.

**To Reproduce**
Steps to reproduce the behavior:

Here is my unit test sequence
 ```
plss_state_det_UT_app.c
    UT_SetDeferredRetcode(UT_KEY(CFE_SB_RcvMsg), 1, CFE_SUCCESS);
    UT_SetDeferredRetcode(UT_KEY(CFE_SB_RcvMsg), 1, !CFE_SUCCESS);      // in ProcessNewCmds()
    UT_SetDeferredRetcode(UT_KEY(CFE_SB_RcvMsg), 1, !CFE_SUCCESS);      // in ProcessNewData()
    UT_SetForceFail(UT_KEY(CFE_SB_GetMsgId), PLSS_STATE_DET_WAKEUP_MID);     // #define 0x18D2   <-----  Setting the MsgID to be returned by CFE_SB_GetMsgId
    UT_SetHookFunction(UT_KEY(CFE_EVS_SendEvent), UTHF_CFE_EVS_SendEvent_HookMyString, NULL);
  
    iRetVal = PLSS_STATE_DET_RcvMsg(5);      // arbitrary iBlocking value, CFE_SB_RcvMsg is being hooked
 ```
```
plss_state_dep_app.c
int32 PLSS_STATE_DET_RcvMsg(int32 iBlocking)
{
…
    iStatus = CFE_SB_RcvMsg(&MsgPtr, g_PLSS_STATE_DET_AppData.SchPipeId, iBlocking);   ß-------------  Returns MsgPtr pointing to Buffer.Msg (set to all zeros)
 
…
    if (iStatus == CFE_SUCCESS)
    {
        MsgId = CFE_SB_GetMsgId(MsgPtr);         <--------  Returns 0, using Buffer.Msg not using StubEntry data  
        switch (MsgId)
       {
            case PLSS_STATE_DET_WAKEUP_MID:
 ```
 
The MsgId is 0 which should be PLSS_STATE_DET_WAKEUP_MID.
The reason is because the MsgPtr points to Buffer.Msg (set to all zeros) defined in CFE_SB_RcvMsg
 
```
cfe/fsw/cfe-core/ut-stubs/ ut_sb_stubs.c:
int32 CFE_SB_RcvMsg(CFE_SB_MsgPtr_t *BufPtr,
                    CFE_SB_PipeId_t PipeId,
                    int32 TimeOut)
{
    int32 status;
    static union
    {
        CFE_SB_Msg_t Msg;
        uint8 Ext[CFE_MISSION_SB_MAX_SB_MSG_SIZE];
    } Buffer;
 
    status = UT_DEFAULT_IMPL(CFE_SB_RcvMsg);
    if (status >= 0)
    {
        if (UT_Stub_CopyToLocal(UT_KEY(CFE_SB_RcvMsg), (uint8*)BufPtr, sizeof(*BufPtr)) < sizeof(*BufPtr))
        {
            memset(&Buffer, 0, sizeof(Buffer));
            *BufPtr = &Buffer.Msg;
``` 
 
The MsgPtr is passed to the CFE_SB_GetMsgId, where the MsgId is calculated using        #define CCSDS_RD_SID(phdr)         (((phdr).StreamId[0] << 8) + ((phdr).StreamId[1]))
The calucated MsgId is 0 since MsgPtr is pointing to Buffer.Msg which is set to all zeros.
 
 ```
CFE_SB_MsgId_t CFE_SB_GetMsgId(const CFE_SB_Msg_t *MsgPtr)
{
    CFE_SB_MsgId_t MsgId = 0;
 
    UT_DEFAULT_IMPL(CFE_SB_GetMsgId);
 
    if (UT_Stub_CopyToLocal(UT_KEY(CFE_SB_GetMsgId), &MsgId, sizeof(MsgId)) < sizeof(MsgId))
    {
#ifdef MESSAGE_FORMAT_IS_CCSDS
 
#ifndef MESSAGE_FORMAT_IS_CCSDS_VER_2 
        MsgId = CCSDS_RD_SID(MsgPtr->Hdr);
#else
 ```
 
Note: For UT_DEFAULT_IMPL(CFE_SB_GetMsgId); I.  I do see PLSS_STATE_DET_WAKEUP_MID  being used.

**Expected behavior**
Return PLSS_STATE_DET_WAKEUP_MID

**System observed on:**
 - Hardware
 - OS: Centos 7
 - Versions:cFE 6.7.12.0, OSAL 5.0.11.0, PSP 1.4.8.0, chksm 9194

**Reporter Info**
Jose Tovias CACI/LZT Technology

jphickey, 2020-05-01:
This is a fundamental issue with the CFE_SB_GetMsgId/CFE_SB_SetMsgId "stub" routines in that they are basically clones of the real function, which is bad/incorrect for UT.  They don't follow the normal paradigm of letting a UT test case force an output value, they read it from the buffer.  This is also even more broken when extended header feature is enabled.

Previous comments: https://github.com/nasa/cFE/issues/636#issuecomment-618414832
Related issue: https://github.com/nasa/cFE/issues/618

I think this is fixable by making the functions more of a proper stub and adding a hook function within the CFE UT to get the backward-compatible behavior.

jphickey, 2020-05-01:
However the current stub should work if you use "SetDataBuffer" to set a CFE_SB_MsgId_t value for CFE_SB_GetMsgId rather than using UT_SetForceFail.

Not that the latter (UT_SetForceFail) is only usable with functions that have integer return values, and CFE_SB_MsgId_t is an abstract value, not necessarily an integer.  (it is today, but may not be in future versions, think if it like pthread_t).

skliper, 2020-05-01:
A goal after wrapping up the core code updates is to “fix” these stubs and transition the cFE stubs and coverage tests to more like the OSAL model.  We’ll need to coordinate the conversion based on dependencies, maybe a “CFE_SIMPLE_STUBS” option or similar.  Really the coverage tests need to focus on the unit under test, not full stack functionality (that's the role of functional tests and not the mocks/stubs coverage test).

jphickey, 2020-05-01:
I have thoughts about how this could be done without needing to add a compile option.  Most dependencies on the existing behavior are in the CFE UT anyway so its in the same repo and therefore easier to coordinate.

skliper, 2020-05-01:
@jphickey I was hoping you would... and looking forward to fixing the CFE UTs.

jtovias, 2020-05-01:
using SetDataBuffer worked. I was able to get the correct MsgId, Thanks. 

Since this has a related  issue: #618 and I have it working, this can be close. 

Would that be ok with you?

closed 2020-05-04 by skliper