nasa/cFE #663: Mismatched Variable Types in Data Structures

dmknutsen, 2020-05-01:
**Describe the bug**
While auditing cFE tlm packets for 64 bit alignment issues, I noticed that we have instances of mismatched variable types in data structures.  This results in data being truncated/corrupted.

Example:
In cfe_es.h::CFE_ES_AppInfo_t
There are multiple addresses (StartAddress, CodeAddress, etc) declared as uint32.  They should be declared as cpuaddr variables - similar to CFE_ES_AppStartParams_t:StartAddress and OS_module_address_t:code_address.  

Another example is the priority, stackSize, and Exception Action variables.  See below for declaration trace:

CFE_ES_ParseFileEntry:
unsigned int Priority
unsigned int StackSize
unsigned int ExceptionAction

CFE_ES_AppCreate:
uint32  Priority
uint32  StackSize
uint32  ExceptionAction

CFE_ES_AppStartParams_t:
uint16  ExceptionAction
uint16  Priority
uint32  StackSize

CFE_ES_AppInfo_t:
uint16   Priority
uint16   ExceptionAction
uint32   StackSize

**Reporter Info**
Dan Knutsen
NASA/Goddard

jphickey, 2020-05-01:
Part of this was actually a design decision at the time 64 bit support was initially added.  There was a desire to _not_ change the TLM/CMD formats at that time, so these fields remained fixed 32 bit in size and the `AddressesAreValid` should be set FALSE to basically just tell the ground system they are all wrong and not to pay attention to them.

That being said, there might have still been one place (in the mempool stuff?) where telemetry did change size.

jphickey, 2020-05-01:
The other aspect is that CFE doesn't ever actually use data structures that are bigger than 32-bit sizes so the size fields remaining as 32-bit was OK.  The emphasis was to build and run on the newer architectures, not necessarily to actually use or manipulate huge data structures with sizes > 4GB.

dmknutsen, 2020-05-01:
Thanks for the info/feedback!  I just double checked and identified the following tlm/cmd that contain 64 bit variables:
CFE_SB_HousekeepingTlm_Payload
CFE_ES_PoolStatsTlm_Payload
CFE_TBL_HouseKeepingTlm_Payload_t
CFE_TBL_TblRegPacket_Payload_t
CFE_ES_SendMemPoolStatsCmd_Payload_t

I think it might be possible that these were updated at some point following the original decision to not modify and CFE_ES_AppInfo_t was missed because it is contained in cfe_es.h - as opposed to cfe_es_msg.h.

jphickey, 2020-05-01:
Yeah, those are likely all pointers / memory addresses right?  That should be a separate design discussion / steering committee topic on the solution to these.  Firstly it should be clarified whether or not the addresses are even needed in the TLM stream in the first place (i.e. what does the ground system actually use that for).

Obviously, receiving a memory address in a CMD (as in CFE_ES_SendMemPoolStatsCmd) and proceeding to dereference it is very dangerous because CFE cannot validate the pointer value wasn't corrupted at some point.   I think there is a separate issue ticket about this?

For this we should consider using a more abstract "handle" instead, that the CFE can validate and get the real address of.  It can stay 32 bits which will be backward compatible with a 32-bit memory address.

Any other CMD/TLM interactions that deal with direct memory addresses should be relegated to an app (e.g. MM, which already does peeks and pokes) due to the inherent risk involved.

skliper, 2020-08-05:
Add to list of issues that need design discussion.

skliper, 2020-08-05:
Topics touched on - possibly mission define typedef for memory address sizes (could be 64 or 32...), discussed config vs just going to 64 bit everywhere...
GetMemPoolStats - this is an offender for taking memory addresses.

astrogeco, 2020-08-05:
[CCB 2020-0805](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-August-05-2020) See comment above and minutes

skliper, 2020-09-22:
Note CFE_TBL_TblRegPacket_Payload_t is unique in that it actually uses cpuaddr.  Not good.

jphickey, 2020-10-29:
In places where CFE is using these values to call OSAL (e.g. stack pointer and priority at least), OSAL should be identifying the right type to use.  Submitted nasa/osal#635 to add a typedef for those items.  Will then update CFE to use the typedef.

jphickey, 2020-10-29:
My other recommendation is to start migrating toward using `cpusize` (or perhaps `size_t`, because it is a standard C type) for internal sizes - as it will match the type resulting from use of e.g. `sizeof()` operator and thereby avoid any conversions.

Types in commands/telemetry will still need to be standardized by CFE to be fixed width - cannot use `cpusize` here.  But #964 already added a separate typedef for those.  So the remaining issue is primarily how it is stored internally and that we do many conversions across APIs.

jphickey, 2020-10-29:
After reviewing the code a bit more I'm leaning more toward using the standard `size_t` internally and using the `CFE_ES_MemOffset_t` for sizes cmd/tlm messages - perhaps renaming it to `CFE_ES_MemOffset_Atom_t` to clarify it is intended for this purpose.

jphickey, 2020-10-29:
Another potential compromise to move in that direction without jumping all-in would be to add another typedef in ES for this, i.e.

    typedef size_t CFE_ES_Size_t;

Then update all CFE APIs to use this typedef for sizes.  Mem Pool code was already done this way in my previous commit, but this can be generalized to the rest of the API.

Upside: Gives users an easy escape if their apps don't compile nicely with 64-bit sizes - they can patch it back to `uint32`.
Downside: Yet another typedef.....

skliper, 2020-10-29:
I vote against another typedef, I'd rather jump all-in.

PR #1003 (2020-11-21): Fix #663, more consistent use of data types
**Describe the contribution**
Define and Use data types more consistently across CFE.

Replace many uses of generic uint16/uint32 with a more purpose-specific typedef. Replace all sizes with `size_t` across the API.

Fixes #663

**Testing performed**
Build and sanity test CFE

**Expected behavior changes**
None, but does affect API.  The API changes should be largely backward compatible, however.

**System(s) tested on**
Ubuntu 20.04, 64-bit

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Syst [... cut]
commit 2c1c905c (2020-11-21): Fix #663, Scrub datatypes across EVS

Fix all use of uint32 for sizes, replace with size_t
Scrub all other types to make sure use is consistent,
using proper typedef whenever available.
commit 2cea88d8 (2020-11-21): Fix #663, Scrub datatypes across TIME

Fix all use of uint32 for sizes, replace with size_t
Scrub all other types to make sure use is consistent,
using proper typedef whenever available.
commit 4f39bee9 (2020-11-21): Fix #663, replace fixed sizes with sizeof()

Rather than directly referencing a constant, prefer to use
the sizeof() operator on the instance or type whenever possible.
closed 2020-12-02 by astrogeco