nasa/cFE #666: Memory alignment issues in TIME (32bit, MCP750, CCSDS_VER_2 config)

skliper, 2020-05-04:
**Is your feature request related to a problem? Please describe.**
```
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/time/cfe_time_tone.c: In function 'CFE_TIME_Tone1HzTask':
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/time/cfe_time_tone.c:1220: warning: cast increases required alignment of target type
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/time/cfe_time_tone.c:1228: warning: cast increases required alignment of target type
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/time/cfe_time_tone.c: In function 'CFE_TIME_Local1HzTask':
/home/jhageman/cFS/cFS-GitHub/cfe/fsw/cfe-core/src/time/cfe_time_tone.c:1437: warning: cast increases required alignment of target type
```

**Describe the solution you'd like**
Force alignment where possible without changing bits on the wire

**Describe alternatives you've considered**
Remove `-Wcast-align` for this build

**Additional context**
Note there are other alignment issues for other configuration options (#313, #314) but they don't show up for MCP750 with CCSDS Version 2 so aren't critical to 6.8.

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-05-05:
As of right now this is the final core code change for 6.8 - @jphickey

jphickey, 2020-05-05:
When I test this I get the same results on extended vs. non-extended headers.  

I would not expect the setting of this config option to change the alignment requirements of packets as most packets do not contain a CFE_SB_MsgId_t directly.

The reason there are just a few warnings in this build is because most of the alignment warnings were resolved in issue #437 and these are simply all that are left.

skliper, 2020-05-05:
> When I test this I get the same results on extended vs. non-extended headers.

Yes, that's why it's an issue we need to address now.  Extended headers only fixes the alignment warning for packets that contain CFE_SB_MsgId_t.

skliper, 2020-05-06:
@jphickey could you clarify what's the hold-up on resolving this issue?  Your comments above make it sound like we are not on the same page.

jphickey, 2020-05-06:
> @jphickey could you clarify what's the hold-up on resolving this issue? Your comments above make it sound like we are not on the same page.

The challenge here was to come up with something that wouldn't just hide the fundamental flaw (i.e. address the actual alignment of the message structure at issue) but also wouldn't affect other messages.  Hopefully the PR that I just submitted will be acceptable.

The fix I'm proposing is split into two commits, the first just makes the `CFE_SB_CmdHdr_t`/`CFE_SB_TlmHdr_t` types correctly aligned for how they are actually used.  Note most message definitions do not use this type - they define the message header as a `uint8` array based on the `sizeof()` this type (i.e. not aligned). 

The second commit changes the header definition only on these particular message types that were generating warnings.  This makes them aligned and fixes the warning.  

Note the same thing _should_ be done for all messages, because using a `uint8[]` array is inherently unsafe/incorrect as it is not aligned for anything.  It is only because most payloads contain a `uint32` somewhere that it isn't getting warned about elsewhere.

jphickey, 2020-05-06:
Also note this PR only fixes warnings within the FSW code.   I'm still seeing several alignment cast warnings in the unit test code on my 32-bit test platform.  Will submit a separate issue about that.

skliper, 2020-05-06:
Concurrence received on applying this pattern on all the messages, we'll manage the impact of tlm changes as part of this cycle.

PR #678 (2020-05-08): Fix #666, alignment of CMD/TLM message definitions
**Describe the contribution**
Define the SB message headers such that they are aligned appropriately for casting to/from a `CFE_SB_Msg_t` type.  Note this should not affect the size of these structures on a 32-bit machine with the default configuration, as they are already multiples of `sizeof(uint32)`....

Then change only the CMD/TLM definitions which were generating warnings on the 32-bit build to use this correct definition, rather than a `uint8[]` array to reserve space for the header.  Again in these cases this should not change the size [... cut]
commit 8e8c4657 (2020-05-08): Merge pull request #678 from jphickey/fix-666-sb-message-alignment

Fix #666, alignment of CMD/TLM message definitions
commit 1d99380d (2020-05-08): Merge pull request #678 from jphickey/fix-666-sb-message-alignment

Fix #666, alignment of CMD/TLM message definitions
PR #692 (2020-05-13): Integration Candidate COMBINED 2020-04-29 and 2020-05-06
**Describe the contribution**

Fix #399, Deprecate decompress
Fix #411, rework exception handling in CFE
Fix #484, Deprecate output to shell commandFix #491, Add UT macros
Fix #519, Deprecate CFS/FS time conversion APIs
Fix #523, SB Subscription report control on separate MID
Fix #551, Remove duplicate CFE_TIME_Local1HzISR prototype
Fix #559, Resolve doxygen warnings
Fix #618, stubs must not depend on real msgid implementation
Fix #637, Remove old CFE_ARINC653-related tests
Fix #639, Remove ES_APP_DEBUG functionality
Fix #641, string ope [... cut]
closed 2020-05-08 by astrogeco