nasa/cFE #675: whole-archive-unrecognized

jtovias, 2020-05-06:
I can work this issue.

**Describe the bug**
In START_WHOLE_ARCHIVE and STOP_WHOLE_ARCHIVE, it is missing "-Wl," when compiling for vxworks. 

I am using a sparc compiler and I get the following error: 

Linking C executable core-plss_cws
cd  ..........    -g    CMakeFiles/core-plss_cws.dir/src/target_config.c.obj  -o core-plss_cws  --whole-archive ../cfe_core_default_plss_cws/libcfe_core_default_plss_cws.a ../psp/ut700-vxworks6.9/libpsp-ut700-vxworks6.9.a ../osal/libosal.a --no-whole-archive 

sparc-wrs-vxworks-gcc: error: unrecognized command line option '--whole-archive'
sparc-wrs-vxworks-gcc: error: unrecognized command line option '--no-whole-archive'

To fix the error see below in code snips:

**To Reproduce**
Steps to reproduce the behavior for XEMU PLSS project:
1. Go to plss_cws_mission
2. source setvars.sh
3. mcrv
4. See error

**Expected behavior**
No error.  A clean build.

**Code snips**
 I updated cafe/cmake/target/CMakeLists.txt

  if ("${CMAKE_C_COMPILER_ID}" STREQUAL "GNU")
    # The option pair for GNU gcc/ld tools
    if(SIMULATION STREQUAL "linux")
       set(START_WHOLE_ARCHIVE "--whole-archive")
       set(STOP_WHOLE_ARCHIVE  "--no-whole-archive")
    else()
       set(START_WHOLE_ARCHIVE "-Wl,--whole-archive")
       set(STOP_WHOLE_ARCHIVE  "-Wl,--no-whole-archive")
     endif()

    # the linker option prefix may or may not be needed, see below
    set(COMPILER_LINKER_OPTION_PREFIX "-Wl,")
  else()
    # Other toolchain options may be added here
  endif()

**System observed on:**
 - Hardware
 - OS: Centos 7
 - Versions:cFE 6.7.12.0, OSAL 5.0.11.0, PSP 1.4.8.0, chksm 9194
 - VxWorks6.9

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Jose Tovias CACI/LZT Technology

jphickey, 2020-05-06:
This is likely because whatever system you are using does not define `CMAKE_EXE_EXPORTS_C_FLAG`

See `psp/cmake/Modules/Platforms/RTEMS.cmake` for how this was done on that platform.

This has come up periodically -- the current approach of trying to use "whole-archive" option is toolchain specific and it would be nice if we could come up with a more platform-independent way of doing this.

skliper, 2020-05-06:
Marking as duplicate with recommendation to follow RTEMS model until we come up with a better solution.

closed 2020-05-06 by skliper

jtovias, 2020-05-06:
I tested using CMAKE_EXE_EXPORTS_C_FLAG, my load compiled without errors. Thanks..