nasa/cFE #680: remove conditional TBL compilation logic

CDKnightNASA, 2020-05-06:
**Is your feature request related to a problem? Please describe.**
With #660 TBL is no longer optional; conditional code/compilation configuration should be removed.

**Requester Info**
Christopher.D.Knight@nasa.gov

skliper, 2020-05-06:
Also remove the conditional description in the doxygen documentation: 
https://github.com/nasa/cFE/blob/8365a0c69b810e313afe0fe19cdc8987374621e3/docs/src/cfe_tbl.dox#L31-L62

jphickey, 2020-10-16:
Note - as part of fixing #950 I'm rearranging some of the cleanup logic, and the call to `CFE_TBL_CleanUpApp()` is currently conditional on `EXCLUDE_CFE_TBL`.  I'd like to drop the `#ifdef` here as part of my upcoming change.

Here is the `#ifdef` I want to remove:
https://github.com/nasa/cFE/blob/dc3d62b37d2bacd8fe23f66ff7b64025ac7d0751/fsw/cfe-core/src/es/cfe_es_apps.c#L1104-L1106

There are two other places it appears in `cfe_es_objtab.c` which would still need to be cleaned up, assuming we go through with this for 7.0

skliper, 2020-11-03:
Note if we update to the services being modules, tbl removal/customization comes almost for free (still need to remove all the EXCLUDE_CFE_TBL occurrences).  Could just make non-ops for the few calls that do exist, or customize/replace whatever is desired and replace the core module.  Related to #972.

PR #1132 (2021-02-04): Fix #680, Remove table service exclusion logic
**Describe the contribution**
Fix #680 - This removes the final remnants of the table service exclusion logic and documentation.

**Testing performed**
Build and run unit tests, passed.

**Expected behavior changes**
No affect to default behavior.  `EXCLUDE_CFE_TBL` no longer used, if define table services will still be started.

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: Bundle main + this commit

**Additional context**
#972 will transition table service to a module that can be customized or implemented d [... cut]
commit 8257c1e0 (2021-02-04): Fix #680, Remove table service exclusion logic
commit e0cb584b (2021-02-04): Merge pull request #1132 from skliper/fix680-conditional_tbl

Fix #680, Remove table service exclusion logic
PR #1150 (2021-02-16): cFE  Integration candidate: 2021-02-02
**Describe the contribution**
Fix #1070, patch PSP include directory reference
Fix #1108, split resource id functions into module
Fix #723, CFE_EVS_Register const correct and report truncation
Fix #680, Remove table service exclusion logic
Fix #609, Remove CFE_PLATFORM_EVS_LOG_ON undefined option to diasble log
Fix #915, Check for file existence in CFE_ES_RestartApp
Fix #1122, ES and EVS use default message limits
Fix #1123, Convert global CFE_SB_Default_Qos to macro
Fix #780, Hex output for return checking ASSERTs
Fix #986, Remove unnecessary CFE_MSG_Ini [... cut]
closed 2021-02-16 by astrogeco