nasa/cFE #687: TblHandle used before validation in CFE_TBL_Load

skliper, 2020-05-07:
**Describe the bug**
TblHandle used before validation

**To Reproduce**
See code snip, invalid handle could cause fault

**Expected behavior**
Validate handle before using

**Code snips**
https://github.com/nasa/cFE/blob/8365a0c69b810e313afe0fe19cdc8987374621e3/fsw/cfe-core/src/tbl/cfe_tbl_api.c#L681-L687

**System observed on:**
In code

**Additional context**
None

**Reporter Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-05-07:
Ping @dmills3

jphickey, 2020-05-07:
This isn't really using the table handle - there is no dereferencing of the pointer before validation, so there is no chance of segfault.  Maybe less than ideal form to (possibly) set a pointer to something invalid, but it is not a fault risk.

jphickey, 2020-05-07:
deleted my previous comment - I see it now ... yes this is a bug.

jphickey, 2020-05-07:
This code was last modified by @CDKnightNASA in #606 so probably should get his take on best fix.

PR #688 (2020-05-19): Fix #687, validate table handle prior to use
**Describe the contribution**
fixes a bug where a table handle is used prior to validation (used in the variable initialization at the top of CFE_TBL_Load() .)

Fix #687

**Contributor Info - All information REQUIRED for consideration of pull request**
Christopher.D.Knight@nasa.gov
commit 0fa7baf1 (2020-05-19): Merge pull request #688 from CDKnightNASA/fix-687-tblhandle_verify

Fix #687, validate table handle prior to use
commit 84809a77 (2020-05-19): fix #687 - validate table handle prior to use
closed 2020-05-22 by astrogeco