nasa/cFE #702: ut_sb_stubs.c needs a stub for CFE_SB_ZeroCopyGetPtr

AlanSGibson, 2020-05-14:
**Is your feature request related to a problem? Please describe.**
CFE_SB_ZeroCopyGetPtr does not have a stub in ut_sb_stubs.c causing undefined reference error when building unit tests for an app that uses this call.

**Describe the solution you'd like**
Add the stub for CFE_SB_ZeroCopyGetPtr with full context tracking.

**Describe alternatives you've considered**
Create my own locally in the app's unit test, but this does not help everyone.

**Requester Info**
Alan Gibson, NASA

skliper, 2020-05-14:
Contribute your implementation?

jphickey, 2020-05-14:
Probably don't need an individual issue ticket for each missing stub though.... maybe just one issue to scrub the public API and make sure we have a stub for each one?

AlanSGibson, 2020-05-14:
I had considered that.  That seems less targeted and more likely to take extra time, whereas these are the ones I currently need, but if you really want that I will close these and create a general issue.

astrogeco, 2020-05-14:
I think it's fine to have small issues. And I think it should be a best practice.

jphickey, 2020-05-14:
Either way is fine but we probably should have a to-do item to scrub the public API for missing stubs, in case one was overlooked elsewhere.

zanzaben, 2021-03-01:
fixed by #763

closed 2021-03-01 by zanzaben