nasa/cFE #719: Remove references to "CCSDS" structures outside of CFE SB.

jphickey, 2020-05-27:
**Is your feature request related to a problem? Please describe.**
As a prerequisite to #554, all modules other than SB itself should refer specifically to the "CCSDS" data types and macros.

**Describe the solution you'd like**
Remove references to the `ccsds.h` header file along with any direct references to types defined in this file.  Use the types or abstractions defined in `cfe_sb.h` instead.

**Additional context**
Mostly an issue for the "Send HK" commands that accept a `CCSDS_CommandPacket_t` structure.  

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2020-05-28:
@jphickey could you prioritize this one?

jphickey, 2020-05-28:
> @jphickey could you prioritize this one?

Submitted PR, will submit similar issues/PRs for apps too.

jphickey, 2020-05-28:
FYI: nasa/ci_lab#49, nasa/to_lab#42, nasa/sample_app#69 are the related PRs for the apps.

PR #725 (2020-06-02): Fix #719, remove refs to ccsds data structures
**Describe the contribution**

Replace all direct references to data types defined in `ccsds.h` with the abstract type defined in `cfe_sb.h`.

Fixes #719 

**Testing performed**
Build and run CFE, sanity check, confirm all unit test pass.

**Expected behavior changes**
No impact to behavior.

**System(s) tested on**
Ubuntu 20.04

**Additional context**
There are similar changes to apps, too, which will be separate PRs (not submitted yet).

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Syst [... cut]
commit 0cad2698 (2020-06-02): Merge pull request #725 from jphickey/fix-719-remove-ccsds-refs

Fix #719, remove refs to ccsds data structures
commit 755851f9 (2020-06-02): Fix #719, remove refs to ccsds data structures

Replace all direct references to data types defined in ccsds.h
with the abstract type defined in cfe_sb.h
PR #729 (2020-06-10): Integration Candidate: 2020-05-27
**Describe the contribution**
Fix #704
Fix #603
Fix #719

**Testing performed**
https://github.com/nasa/cFS/pull/96/checks

**Expected behavior changes**
PR #714 -  API Change: cFE ut_sb_stubs now has CFE_SB_DeletePipe available.
    Behavior Change: App unit tests requiring this will not fail to build due to undefined reference to `CFE_SB_DeletePipe`

PR #713 -  Hook functions may now use the `va_list` form and obtain the full set of variable arguments passed to `CFE_EVS_SendEvent` and variants. 

PR #725 - Replace all direct references to data types defined  [... cut]
closed 2020-06-10 by astrogeco