nasa/cFE #727: multiple targets - not able to see cfe_msgids.h when building tables

CDKnightNASA, 2020-06-02:
**Describe the bug**
When building for multiple targets, app tables fail to build, reporting "cfe_msgids.h: No such file or directory".
 
**To Reproduce**
```
git clone https://github.com/nasa/cFS
cd cFS
git submodule init
git submodule update
cp cfe/cmake/Makefile.sample Makefile
cp -r cfe/cmake/sample_defs .
sed -e 's/#SET(TGT2/SET(TGT2/' < .cfe/cmake/sample_defs/targets.cmake > sample_defs/targets.cmake
echo '!' > sample_defs/cpu2_cfe_es_startup.scr
cp sample_defs/cpu1_msgids.h sample_defs/cpu2_msgids.h
sed -e 's/CPU_ID 1/CPU_ID 2/' < sample_defs/cpu1_platform_cfg.h | sed -e 's/CPU_NAME "CPU1"/CPU_NAME "CPU2"/' > sample_defs/cpu2_platform_cfg.h
export SIMULATION=native
make
```

**Expected behavior**
Builds.

**System observed on:**
Debian 10.3 VM

**Reporter Info**
Christopher.D.Knight@nasa.gov

CDKnightNASA, 2020-06-02:
make output:

```
mkdir -p "build"
(cd "build/" && cmake  -DCMAKE_INSTALL_PREFIX=/exe -DCMAKE_BUILD_TYPE=debug "/home/user/cFS/cfe")
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Mission configuration sample automatically selected
-- OMIT_DEPRECATED=false: Deprecated elements included in build
-- Module 'sample_app' found at /home/user/cFS/apps/sample_app
-- Module 'sample_lib' found at /home/user/cFS/apps/sample_lib
-- Module 'ci_lab' found at /home/user/cFS/apps/ci_lab
-- Module 'to_lab' found at /home/user/cFS/apps/to_lab
-- Module 'sch_lab' found at /home/user/cFS/apps/sch_lab
-- Module 'cfe-core' found at /home/user/cFS/cfe/fsw/cfe-core
-- Module 'osal' found at /home/user/cFS/osal
-- Configuring for system arch: native
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- OMIT_DEPRECATED=false: Deprecated elements included in build
-- BSP Selection: pc-linux at /home/user/cFS/osal/src/bsp/pc-linux
-- OSAL Compile Definitions: SIMULATION=native;_LINUX_OS_;_XOPEN_SOURCE=600
-- OSAL Selection: posix at /home/user/cFS/osal/src/os/posix
-- PSP Selection: pc-linux
-- Building App: sample_app install=cpu1;cpu2
NOTE: Selected /home/user/cFS/apps/sample_app/fsw/src/sample_table.c as source for sample_table
NOTE: Selected /home/user/cFS/apps/sample_app/fsw/src/sample_table.c as source for sample_table
-- Building App: sample_lib install=cpu1
-- Building App: ci_lab install=cpu1;cpu2
-- Building App: to_lab install=cpu1;cpu2
-- Building App: sch_lab install=cpu1;cpu2
NOTE: Selected /home/user/cFS/apps/sch_lab/fsw/src/sch_lab_table.c as source for sch_lab_table
NOTE: Selected /home/user/cFS/apps/sch_lab/fsw/src/sch_lab_table.c as source for sch_lab_table
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/cFS/build/native
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/cFS/build
echo " -DCMAKE_INSTALL_PREFIX=/exe -DCMAKE_BUILD_TYPE=debug" > "build/.prep"
make --no-print-directory -C "build" mission-all
Scanning dependencies of target elf2cfetbl
[ 50%] Building C object tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o
[100%] Linking C executable elf2cfetbl
[100%] Built target elf2cfetbl
Scanning dependencies of target mission-version
[100%] Built target mission-version
Scanning dependencies of target mission-prebuild
[100%] Built target mission-prebuild
Scanning dependencies of target native-all
Scanning dependencies of target osal_posix_impl
[  1%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-binsem.c.o
[  1%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-common.c.o
[  2%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-console.c.o
[  3%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-countsem.c.o
[  3%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-dirs.c.o
[  4%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-errors.c.o
[  4%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-files.c.o
[  5%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-filesys.c.o
[  5%] Building C object osal/posix_impl/CMakeFiles/osal_posix_impl.dir/src/os-impl-heap.c.o
[... 87 lines cut ...]
[ 47%] Built target sample_app
Scanning dependencies of target sample_lib
[ 48%] Building C object apps/sample_lib/CMakeFiles/sample_lib.dir/fsw/src/sample_lib.c.o
[ 48%] Linking C shared module sample_lib.so
[ 48%] Built target sample_lib
Scanning dependencies of target ci_lab
[ 50%] Building C object apps/ci_lab/CMakeFiles/ci_lab.dir/fsw/src/ci_lab_app.c.o
[ 50%] Linking C shared module ci_lab.so
[ 50%] Built target ci_lab
Scanning dependencies of target to_lab
[ 51%] Building C object apps/to_lab/CMakeFiles/to_lab.dir/fsw/src/to_lab_app.c.o
In file included from /home/user/cFS/apps/to_lab/fsw/src/to_lab_app.c:73:
/home/user/cFS/apps/to_lab/fsw/platform_inc/to_lab_sub_table.h:33:10: fatal error: cfe_msgids.h: No such file or directory
 #include "cfe_msgids.h"
          ^~~~~~~~~~~~~~
compilation terminated.
make[7]: *** [apps/to_lab/CMakeFiles/to_lab.dir/build.make:63: apps/to_lab/CMakeFiles/to_lab.dir/fsw/src/to_lab_app.c.o] Error 1
make[6]: *** [CMakeFiles/Makefile2:670: apps/to_lab/CMakeFiles/to_lab.dir/all] Error 2
make[5]: *** [Makefile:141: all] Error 2
make[4]: *** [CMakeFiles/native-all.dir/build.make:57: CMakeFiles/native-all] Error 2
make[3]: *** [CMakeFiles/Makefile2:330: CMakeFiles/native-all.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:144: CMakeFiles/mission-all.dir/rule] Error 2
make[1]: *** [Makefile:201: mission-all] Error 2
make: *** [Makefile:120: all] Error 2
```

jphickey, 2020-06-02:
This is actually a duplicate of issue #724 -- specifically you are seeing the third issue in my comment here: https://github.com/nasa/cFE/issues/724#issuecomment-635549646

The "quirk" is that once you move to two cpus that share a platform (native) you no longer have access to the cfe config, and the cfe_msgid.h file is part of that (same as cfe_platform_cfg.h).

jphickey, 2020-06-02:
Note in this case its also because to_lab violates the rules and it doesn't actually build its subscription list as a table.  I believe the that the actual table build process _would_ have had access to this file OK.  But this compiles the table as a regular source file.

astrogeco, 2020-06-02:
I hesitate to call this a duplicate. Would it make sense to split up #724 which seems to be fairly big right now? Or can this multiple cpu issue be addressed without the overhauls proposed in #724 ?

jphickey, 2020-06-02:
It's different symptoms of one problem.  There is already an issue about the way to_lab does its non-table (see nasa/to_lab#32).  If that issue were fixed and it became a real table, it would probably mask/hide this issue.

I actually have a set of changes I was already assembling into a PR - it should resolve this.  It basically makes it so each CPU is treated as a different build by default unless you specify TGTx_PLATFORM in your targets.cmake file.  It's a bit of a back-step as far as modularity goes but most likely we can't go fullly toward reusable app binaries until we have EDS or something like it.  The existing paradigm of compiling-in your msg IDs which are different for each CPU kinda breaks it.

CDKnightNASA, 2020-06-02:
ok I'll consider this closed and look to fix to_lab (which could use some TLC anyways. :) )

closed 2020-06-02 by CDKnightNASA

jphickey, 2020-06-02:
@CDKnightNASA -- See PR #728.  Can you pull this into your build and see if it behaves more like you expected?

CDKnightNASA, 2020-06-02:
> @CDKnightNASA -- See PR #728. Can you pull this into your build and see if it behaves more like you expected?

Yes that PR fixes the problem, thanks and looking forward to approving it for merge into master.