nasa/cFE #730: low CFE_MISSION_MAX_FILE_LEN and CFE_TBL_FILEDEF(..., "sample_app_table.tbl") causes silent non-truncation

CDKnightNASA, 2020-06-03:
**Describe the bug**
I was scratching my head why my table file was named "sample_app_table.tbl\004"  until I realized that CFE_MISSION_MAX_FILE_LEN is defined as "20" and "sample_app_table.tbl" is exactly 20 characters. If I had one more char, the compiler would err. But it accepts static definition of structs using strings that are the exact length, which loses the null terminator.

**Expected behavior**
Should allow file names of CFE_MISSION_MAX_FILE_LEN chars.

**Reporter Info**
Christopher.D.Knight@nasa.gov

CDKnightNASA, 2020-06-05:
I've pushed an alternative approach of tacking on "\0" to force a compile error if the strlen is the size of the array. Thoughts?

ericgilligan-nasa, 2020-06-11:
I ran into this exact same issue today. Was scratching my head for a long time on it

CDKnightNASA, 2020-06-11:
> I ran into this exact same issue today. Was scratching my head for a long time on it

Then you can test out my pull request :D (I seriously forgot to create it.) Note that it's PR #738 that you want to test out...The other PR linked to this was a previous idea that I scrapped, and had accidentally merged in a different branch into the PR branch. Whoop.

PR #738 (2020-06-23): fix #730 - adds null to FILEDEFs
**Describe the contribution**
Fix #730 - Adds "\0" in the CFE_TBL_FILEDEF() macro so that if a string is exactly the size of the field, it will produce a compiler error.

**Contributor Info - All information REQUIRED for consideration of pull request**
Christopher.D.Knight@nasa.gov
commit c37197ea (2020-06-23): fix #730 - adds null to FILEDEFs
PR #756 (2020-06-25): Integration Candidate: 2020-06-17
**Describe the contribution**
Fix #730
Fix #604
Fix #754

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/106/checks

**Expected behavior changes**

PR #738 - Adds "\0" in the CFE_TBL_FILEDEF() macro so that if a string is exactly the size of the field, it will produce a compiler error.

PR #742 - Added cFE User's Guide Reference to README.md

PR #755 - Removes old license

**System(s) tested on**
Ubuntu: Bionic

**Additional context**
Part of https://github.com/nasa/cFS/pull/106

**Third party code**
None

**Contributor Info - All informati [... cut]
commit d8e09b52 (2020-07-01): fix #730 - adds null to FILEDEFs
closed 2020-06-25 by astrogeco