nasa/cFE #736: Version 2 MsgId construction doesn't match description, overloads bits

skliper, 2020-06-06:
**Describe the bug**
Version 2 code takes the full APID (0x3FF mask), or's in a bit for cmd/tlm (0x80 mask) then or's in the Subsystem ID shifted by 8

That means if a user defines an APID of 0x80 for a telemetry message (which is valid per CCSDS), the system will report it as type cmd if it gets the type from the msgid.  It's also a collision between 0x3 bits from the Subsystem ID and the 0x300 bits of APID.

Basically logic doesn't mirror:
CFE_SB_SetMsgId of 0x3FF -> APID = 0x7F, Type = Cmd, SubsystemID = 3
CFE_SB_GetMsgId from APID=0x3FF, Type = Tlm, SubsytemID = 0 -> MsgId = 0x3FF

**To Reproduce**
N/A - code inspection

**Expected behavior**
Get/Set should mirror (SetMsgId should not overload bits)

**Code snips**
https://github.com/nasa/cFE/blob/d217ca39f1de74695f0d4e0eb427f90fbe8b1156/fsw/cfe-core/src/sb/cfe_sb_msg_id_util.c#L118-L152
https://github.com/nasa/cFE/blob/d217ca39f1de74695f0d4e0eb427f90fbe8b1156/fsw/cfe-core/src/sb/cfe_sb_msg_id_util.c#L155-L187
https://github.com/nasa/cFE/blob/d217ca39f1de74695f0d4e0eb427f90fbe8b1156/fsw/cfe-core/src/sb/cfe_sb_msg_id_util.h#L64-L75

**System observed on:**
N/A

**Additional context**
Uncovered as part of #711 work

**Reporter Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-06-08:
I noticed the same thing earlier.... recommendation is to keep the MsgId bits the same between v1 and v2.  No reason to move them, really.  

It makes for weird packing but one can take the full 11 bits of APID into positions [0-10], then pack the CMD/TLM bit into bit position 12. Then 4 bits of subsystem ID go into bits 11, [13-15].  This keeps the CMD/TLM bit in the same place as v1.  Example in PR #716.

jphickey, 2020-06-08:
If we move the CMD/TLM bit, then it risks breaking apps that treat it as an integer and look at bits directly.  It also makes all of the hardcoded example MID values all apps (CI_LAB, TO_LAB, SAMPLE_APP) incompatible with v2 deployment.

skliper, 2020-06-08:
I'm just going to fix the overload as part of #726.  The PR will also make it easier for missions to reassign the MsgId bits however they want, so there is no need to change the framework to guess at what mission requirements are.

jphickey, 2020-06-08:
Well I was under the impression that one of goals of this whole effort was to remove the need for something like `MESSAGE_FORMAT_IS_CCSDS_VER_2` mission config option.  This means either CFE_SB_MsgId_t has to be consistently defined (header type agnostic) or that the definition is actually part of the modular library.  I don't think the latter is appropriate as it is supposed to be an SB abstraction.  But it could work either way.  The main issue I had was with moving bits around based on `MESSAGE_FORMAT_IS_CCSDS_VER_2` isn't really scalable to more options.

skliper, 2020-06-08:
Right, I'd rather deprecate the V2 msgid from within the framework and just support the original definition by default.  Missions can redefine as needed (and update all the hardcoded values and groundsystem to match).  I'm not convinced we should remove the CCSDS extended header inclusion option (but update to use source selection via CMake eventually), since it's standards based and likely useful to support out of the box.

jphickey, 2020-06-08:
It just has to be clearly defined as to what entity "owns" the `CFE_SB_MsgId_t` type definition.  The choices are CFE_SB itself, or the MSG library which is implementing the headers (potentially user-supplied).

My recommendation is to keep it owned by CFE_SB itself because many of its API calls use it and the whole point of its existence is to provide an abstract, implementation-independent definition of a software bus endpoint.  If the definition has to change based on the header implementation then it's not really an abstraction and there is not much point to its existence.

skliper, 2020-06-08:
I moved it to MSG since it depends on specific bits in the header.  Basically anything directly accessing bits conceptually is MSG.  I'd think of it as the layer that DOES change based on header implementation such that SB doesn't have to.

jphickey, 2020-06-08:
OK - but that means that every function that queries/manipulates a CFE_SB_MsgId_t value also needs to move to the MSG library with it....

PR #726 (2020-08-20): Fix #711, Separate secondary header access functions
**Describe the contribution**
Fix #711 - Separates header manipulation and definitions for easier mission customization 
Fix #733 - Fixes validate checksum description to match implementation
Fix #736 - Fixes get msgid logic to not override bits
Fix #597 - Removes local endian SID macros
Fix #529 - By adding an API that does support maximum msg size reporting, note also fixes limit in CFE_SB_InitMsg by the same means.
Fix #781 - Fixes set/get message id logic to not override bits

 - Enables source selection and out-of-tree mission defined o [... cut]
commit d3a073e9 (2020-08-20): Fix #711, Add message header module

Implements message header module such that users
can customize or extend as needed.
 - Separates code for easier selection
 - Implements consistent getter/setter APIs
 - Fix #736/#781: MsgId logic no longer overrides bits in init
 - Fix #529: Get size supports max size
 - Adds single big endian time implementation, but not selected
 - Adds msg module to module list
 - Adds msg module to cppcheck
PR #833 (2020-08-25): Integration Candidate: 2020-08-19
**Describe the contribution**
Fix #794 
Fix #795 
Fix #92 
Fix #711 
Fix #733 
Fix #736 
Fix #597 
Fix #529 
Fix #781 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/130/checks

**Expected behavior changes**
PR #819 - Add msg stubs, update `SB_UT` to use them, and remove msg module include from unit tests

PR #801 - Collapses time options down to just 32 bit second, 16 bit subsecond, always big endian. Removes old defines, and triggers an error if the configuration is set to a format that was removed.

PR #726 - Enables source selection an [... cut]
closed 2020-08-25 by astrogeco