nasa/cFE #741: Should an app call CFE_TBL_Unregister?

pcooksey, 2020-06-12:
**Describe the bug**
Should an app call CFE_TBL_Unregister?
On CFE_ES_ExitApp I get:
```
1980-012-14:03:21.25234 CFE_TBL:RemoveAccessLink-PutPoolBuf[0] Fail Stat=0xC4000007, Hndl=0x56709730, Buf=0x00000000
1980-012-14:03:21.25236 CFE_TBL:RemoveAccessLink-PutPoolBuf[0] Fail Stat=0xC4000007, Hndl=0x56709730, Buf=0x00000000
```

The [documentation](https://github.com/nasa/cFE/blob/master/docs/cFE%20Application%20Developers%20Guide.md#82-registering-tables) no longer says anything about it so I'm starting to think it shouldn't be used. The [sample_app](https://github.com/nasa/sample_app/blob/4a4b8e281802f244a2facdcb3c32238751c678f3/fsw/src/sample_app.c) in the cFS repo also doesn't call CFE_TBL_Unregister either.

The code seems to say we should though:
https://github.com/nasa/cFE/blob/95f34d25cb2843b8ba9db6338bb0b53cb3e38f92/fsw/cfe-core/src/inc/cfe_tbl.h#L334-L339
 
**To Reproduce**
1. Register table
2. Unregister table
3. Call CFE_ES_ExitApp

**Expected behavior**
No errors on exit

**Code snips**
N/A

**System observed on:**
 - Hardware: Laptop
 - OS: Linux
 - Versions: Versions:cFE 6.7.7.0, OSAL 5.0.6.0, PSP 1.4.4.0

**Additional context**
N/A

**Reporter Info**
Philip Cooksey, NASA Ames

pcooksey, 2020-06-17:
It seems by default the tables are being cleaned up automatically as shown below which is part of the CFE_ES_CleanUpApp function:
https://github.com/nasa/cFE/blob/d217ca39f1de74695f0d4e0eb427f90fbe8b1156/fsw/cfe-core/src/es/cfe_es_apps.c#L1083-L1085

This clean up call eventually calls:
https://github.com/nasa/cFE/blob/95f34d25cb2843b8ba9db6338bb0b53cb3e38f92/fsw/cfe-core/src/tbl/cfe_tbl_internal.c#L1429

Which is the same function in CFE_TBL_Unregister:
https://github.com/nasa/cFE/blob/d217ca39f1de74695f0d4e0eb427f90fbe8b1156/fsw/cfe-core/src/tbl/cfe_tbl_api.c#L641

Seems like a bug since an already unregistered table shouldn't be cleaned up again by CFE_ES_CleanUpApp -> CFE_TBL_CleanUpApp.

skliper, 2020-06-17:
> Seems like a bug since an already unregistered table shouldn't be cleaned up again by CFE_ES_CleanUpApp -> CFE_TBL_CleanUpApp.

Agreed.  Updated title and marked as bug.

jphickey, 2021-01-25:
Tested this today - It may have been fixed by other changes, possibly the transition to ES background task and the better locking ES that was implemented with that.

To test - what I did was modify SAMPLE_APP (which registers a table) to only run for 5 seconds, then exit.  As part of exiting, I added a call to `CFE_TBL_Unregister()` before it calls `CFE_ES_ExitApp()` ... I confirmed that the first call to CFE_TBL_Unregister is indeed doing the un-registration, followed by ES invoking `CFE_TBL_CleanUpApp()` via the background task - which did nothing in my test.

The only (minor) issue I observed is that there is an extra "unknown state" event after the app exits itself:
`EVS Port1 66/1/CFE_ES 48: ES_ProcControlReq: Unknown State Application SAMPLE_APP ( 1 )`

But otherwise everything seemed to work just fine.

jphickey, 2021-01-25:
@pcooksey can you check if this issue still exists in the current main branch?

skliper, 2021-01-25:
If it's fixed we should still fix the documentation, table will be unregistered during cleanup so this is only needed if the App wants to unregister explicitly.

PR #1602 (2021-06-10): Fix #741, Document CFE_TBL_Unregister use-case
**Describe the contribution**
Fix #741 - Documents CFE_TBL_Unregister on shared table use case vs use by owning task.

**Testing performed**
CI - Documentation only

**Expected behavior changes**
None

**System(s) tested on**
CI

**Additional context**
#1509 documents potential races, global table handles often aren't protected so care needs to be taken when using shared tables.

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
commit 474c0ff9 (2021-06-10): Merge pull request #1602 from skliper/fix741-tbl_unregsiter_comments

Fix #741, Document CFE_TBL_Unregister use-case
commit d8d2b889 (2021-06-10): Fix #741, Document CFE_TBL_Unregister use-case
PR #1619 (2021-06-16): cFE Integration candidate: 2021-06-15
## Testing

cFE Checks <https://github.com/nasa/cFE/pull/1619/checks> 
- Documentation workflow failures due to osal docs rename. These are fixed in bundle.
- Local tests failure due to osal docs rename, also fixed in bundle. 

cFS Bundle Checks <https://github.com/nasa/cFS/pull/265/checks> 

## Context

Part of https://github.com/nasa/cFS/pull/265

## Authors

@jphickey 
@zanzaben 
@skliper 
@pepepr08 

## Description

### PR #1590 
Fix #1483, Add CI workflow to run cFE coverage tests

### PR #1598 
Fix #1299, Updated FS Read/WriteHeader API return docume [... cut]
closed 2021-06-16 by astrogeco