nasa/cFE #744: SB Pipe Info 'LastSender' Field is Unused

dmccomas, 2020-06-17:
**Describe the bug**
The LastSender field in in the file produced by a SB 'Send Pipe Info to file' command is always zero. After doing some seraches it appears that this field defined in CFE_SB_PipeD_t in cfe_sb_priv.h is never set.

**To Reproduce**
1. Send a SB CFE_SB_SEND_PIPE_INFO_CC command
2. Transfer the file to the ground
3. Decode the binary file 

**Expected behavior**
The structure definition does not have any comments as to what to expect in this field. I would expect the AppId of the of the sender of last message send to the particular pipe.

**Code snips**
n/a

**System observed on:**
 Wndows VM running Ubuntu 18.04
 cfE 6.71, OSAL 5.0.1.0, PSP 1.4.0.0

**Additional context**
None

**Reporter Info**
David McComas, NASA Emeritus

jwilmot, 2020-06-17:
Is the just the reporting or does CFE_SB_GetLastSenderId API also not work?

dmccomas, 2020-06-17:
Just the reporting. It looks like the LastSender field is never referenced
by anyone.  CFE_SB_GetLastSenderId doesn't reference that field.

CDKnightNASA, 2020-06-18:
So cfe_sb.h defines:
```
typedef struct {
    uint32  ProcessorId;/**< \brief Processor Id from which the message was sent */
    char    AppName[OS_MAX_API_NAME];/**< \brief Application that sent the message */
} CFE_SB_SenderId_t;
```

The definition in cfe_sb_priv.h does not define what the LastSender field should be, but I'm guessing ProcessorId is not all that helpful. (Even SBN-relayed messages would show up with the local ProcessorId.)

I will implement as AppId as that seems the most sensible. I'll also add AppId to CFE_SB_SenderId_t as well, and at some point it might make more sense to deprecate AppName from this struct. Note that SBN uses the GetLastSenderId() API to ensure that it doesn't end up with an echo problem.

skliper, 2020-09-09:
This is OBE by #784 which deprecates it.

closed 2020-09-09 by skliper