nasa/cFE #745: CFE_SB_GetLastSenderId() returns a pointer to a shared data location

dmccomas, 2020-06-17:
**Is your feature request related to a problem? Please describe.**
CFE_SB_GetLastSenderId() in cfe_sb_api.c returns a pointer to shared data location which is risky. The function description also warns of some unpredictable or at least inconsistent behavior. 

**Describe the solution you'd like**
I think the use cases for this function should be identified and a safer & behaviorally consistent implementation should be implemented. 
Are the use cases?
1.  Determine if app X send the last message 
2.  Determine whether the last message sent is from a lsit of apps

A couple of options are
1. Caller passes in the name of an app or a list of apps to SB determines whether the last message sent is a match.
2. The current type implementation copies the last message information rather than returning a pointer. 

**Additional context**
n/a

**Requester Info**
David McComas, NASA Emeritus

jwilmot, 2020-06-17:
Use case is to disregard messages from applications that are not "allowed" to send you commands or telemetry.  Example, science app can not send GN&C app a command to turn on/off the thrusters.

Yes, returning a ptr to the global is insecure. That area should be protected.  Option 2 would be my preferred approach as the comparison logic is in the applications domain. SB does not have to lock the area while doing the search across a potentially long list.

CDKnightNASA, 2020-06-18:
I prefer the API return (via an "out" ptr parameter) the AppId of the app. If the caller needs to know the name, there should be a separate call to crosswalk the ID to a name.

CDKnightNASA, 2020-06-18:
Note that my proposed solution requires first changing the CFE_SB_SenderId_t type to include `uint32 AppId`, or replacing SenderId with AppId. Note also ticket #744 which is closely related and should be sequenced with this one lest there be conflicts.

CDKnightNASA, 2020-06-18:
Also note that the documentation of the API in cfe_sb.h describe the return value as the AppId when, if you look at the code, it returns a CFE status code (like CFE_SUCCESS). What a mess. :D

CDKnightNASA, 2020-06-24:
@dmccomas please look at #759, I propose closing this and #744.

skliper, 2020-09-09:
This is OBE by #784 which deprecates it. Marking as duplicate of #759. Merge to main was in #816.

closed 2020-09-09 by skliper