nasa/cFE #753: Secondary Header: extracting wrong bits for command on 32-bit ARM platform

arvrml, 2020-06-22:
The tools/..../cmdUtil has a test to-enable-tlm.sh command to test commands send to a running cFS session   #./core-cpu1.
The command code extracted through the function, CFE_SB_GetCmdCode()  ... eventually calls the macro, CCSDS_RD_FC(...).  This macro masks and shifts bits to extract the command code; however, the cfe/fsw/cfe-core/src/inc/ccsds.h need the following minor modification.

original

#define CCSDS_RD_FC(shdr)           CCSDS_RD_BITS((shdr).Command, 0x7F00, 8)

my change

#define CCSDS_RD_FC(shdr)           CCSDS_RD_BITS((shdr).Command, 0x007F, 0)

Once this code change is made, no problems.

If this is a bug then the masking and bit shifting for all the secondary header should be changed

else please let me know what I might be doing wrong. 

Platform

Beaglebone :  Arm 7 32-bit
Linux beaglebone 4.19.94-ti-r42 #1buster SMP PREEMPT Tue Mar 31 19:38:29 UTC 2020 armv7l GNU/Linux

skliper, 2020-06-22:
Sounds like a versioning issue.  The secondary header was made endian agnostic in the last round of development.  It would help if you followed the provided issue template and filled in all the information.  Specifically, what version of cFE are you using and what version of cmdUtil?

arvrml, 2020-06-23:
Started again:  git clone ...., but this time I stayed with the **master** branch instead of checkout **rc-6.7.0**, the cfe/fsw/cfe-core/src/inc/ccsds.h were set correctly,  so yes, it looks like versioning issue.

However,  issuing  make in folder ~/cFS,  this error appears

...
[100%] Built target core-cpu1
[100%] Built target native-all
[ 14%] Building C object tools/cFS-GroundSystem/Subsystems/cmdUtil/CMakeFiles/cmdUtil.dir/SendUdp.c.o
/home/debian/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/SendUdp.c: In function ‘SendUdp’:
/home/debian/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/SendUdp.c:94:23: error: cast increases required alignment of target type [-Werror=cast-align]
            inet_ntoa(((struct sockaddr_in *)result->ai_addr)->sin_addr), port);

I could work around it but do you have any suggestions?

skliper, 2020-06-23:
Error above is a duplicate of nasa/cFS-GroundSystem#93.  There are a couple suggestions for resolution, but I haven't seen any contributions to resolve it yet.  Closing this issue as duplicate.

closed 2020-06-23 by skliper