nasa/cFE #760: Document the cmake build system

astrogeco, 2020-06-24:
**Describe the bug**
Our documentation does not explain to users how the build system works and how to customize it for their needs

**Expected behavior**
A clear description and possibly some graphics showing how the different cmakelists files link with each other and how dependencies flow.

It might make more sense to move this into the github.com/nasa/cFS instead

**Additional context**
Related to conversation started in https://github.com/nasa/cFE/pull/751 

**Reporter Info**
Gerardo E. Cruz-Ortiz

astrogeco, 2020-06-24:
@JandlynBentley-at-NASA this might be a good one for you to tackle

CraigKnoblauch, 2021-01-10:
I'd like to second the need for this. I'm having a heck of a time figuring out what's needed to customize a build to my needs. @astrogeco What docs do your recommend right now that I could use?

astrogeco, 2021-01-11:
> I'd like to second the need for this. I'm having a heck of a time figuring out what's needed to customize a build to my needs. @astrogeco What docs do your recommend right now that I could use?

I'll have to defer to @jphickey since I can't think of anything other than the comments in the files inside `cmake/sample_defs`

CraigKnoblauch, 2021-01-11:
OK great! In the meantime I can look over those comments.

skliper, 2021-01-11:
@CraigKnoblauch - feel free to post a question here if there is something in particular you are struggling with.  There's a handful of issues that I've marked as "questions" where specific behavior was explained, and getting a better sense of where the challenges are would help with identifying specific documentation needs.

CraigKnoblauch, 2021-01-25:
@astrogeco @skliper I remember a version of cFS that had a setvars.sh script. Where is that file now? Or a better question, has that file's functionality been replaced by something else? Also, where can I find information about how I should set up my mission directory? It's obvious that I should have one, as a vanilla build is complaining about that, but I'm not sure where to start setting it up and telling the build system about it. Do you guys have any ideas?

skliper, 2021-01-25:
`setvars.sh` was used in the "classic build" system before we transitioned to CMake.  It is no longer necessary, and has been replaced by CMake.  I'm not sure what you are asking about "mission directory", do you mean customizing "sample_defs" for your mission?  If so I recommend starting from the "sample_defs" using the quick start steps documented in the https://github.com/nasa/cFS README, and replacing/updating whatever you need to for your mission.

renn0xtek9, 2022-06-05:
Being new to this project and external, I agree that some doc on how the CMake ist structure would be very helpful.
Right now, it is not clear to me why the `cfe/CMakeLists.txt` is used recursively. (I don't grasp why it has to be so)
I have seen this strategy (to deal with multiple platform) in a few project before, but it turned out to be an anti pattern. It tend to confuse people and IDE. @astrogeco do you have some more information on this ?