nasa/cFE #783: ES - command to load a new application, commands to load/unload/reload libraries

CDKnightNASA, 2020-07-27:
**Is your feature request related to a problem? Please describe.**
Currently ES can be commanded to unload or reload an existing application. Commands should be added to load an application not currently loaded. Similarly, there's no commands to load/unload/reload libraries--these commands should be added.

**Describe the solution you'd like**
Addl. commands.

**Describe alternatives you've considered**
SBN currently directly loads libraries, if the library is not already resident. With this change, SBN should command ES to load those libraries. SBN also reloads libraries when the SBN configuration table is changed, this should also translate to ES commands.

**Requester Info**
Christopher.D.Knight@nasa.gov

jphickey, 2020-07-29:
Concur that the fact there is no command or API to [re-]load a library seems like a glaring omission in ES, if we support it for apps, we should support it for libs too.  The only reason I can think of is that it is hard to ensure that a library is not in use - one would have to first stop all apps that use the library, then unload the library, then reload it.  At that point maybe a processor restart would've gotten the job done much faster?

skliper, 2020-07-29:
I seem to recall this has been a topic in the past... and the consensus was it's safer to restart since unloading could break things.

skliper, 2020-07-29:
@acudmore @jwilmot - any background?

CDKnightNASA, 2020-07-29:
> I seem to recall this has been a topic in the past... and the consensus was it's safer to restart since unloading could break things.

Personally, if "load" is unsafe, then so is "unload" and "reload". Either we should support all three in ES or none (and have ES perform the loads privately.) Similar for libraries--while they don't have an associated task, they still have symbols in the symbol table that unloading/reloading could corrupt.

Let's discuss at the CCB today?

skliper, 2020-07-29:
Yes lets!  I realize I didn't provide enough context around my comment.  It was with respect to unloading libraries.

astrogeco, 2020-07-29:
CCB 2020-07-29, More discussion needed schedule a splinter with architects and other stakeholders

iamthad, 2020-07-29:
I wasn't able to make it to today's CCB, but please include me in this splinter. I've spent a good amount of time diving into the details of this type of dynamic loading.

skliper, 2021-01-11:
@dmknutsen - until we get requirements to load/unload/reload libraries, it is not supported.  This has been an ongoing debate for a while, and a general implementation (from ES) is difficult to make "safe".  Leaving this to apps to manage for now.