nasa/cFE #790: Allow target name subdirectory for install files and handle symlinks

excaliburtb, 2020-08-04:
**Is your feature request related to a problem? Please describe.**
Just need to add to the paths of locations for the install list from TGT<N>_FILELIST the possibility of target_defs/<TGT_NAME>/<FILENAME>

Also, if the file is a symlink, it needs to be followed to its source but copied as the name in FILELIST suggests.

**Describe the solution you'd like**
```
    foreach(INSTFILE ${TGT${TGTID}_FILELIST})
      if(EXISTS ${MISSION_DEFS}/${TGTNAME}/${INSTFILE})
        set(FILESRC ${MISSION_DEFS}/${TGTNAME}/${INSTFILE})
      elseif(EXISTS ${MISSION_DEFS}/${TGTNAME}_${INSTFILE})
        set(FILESRC ${MISSION_DEFS}/${TGTNAME}_${INSTFILE})
      elseif(EXISTS ${MISSION_DEFS}/${INSTFILE})
        set(FILESRC ${MISSION_DEFS}/${INSTFILE})
      else()
        set(FILESRC)
      endif()
      if (FILESRC)
        get_filename_component(_resolvedFile "${FILESRC}" REALPATH)
        install(FILES ${_resolvedFile} DESTINATION ${TGT_NAME}/${INSTALL_SUBDIR} RENAME ${INSTFILE})
      else(FILESRC)
        message("WARNING: Install file ${INSTFILE} for ${TGTNAME} not found")
      endif (FILESRC)
    endforeach(INSTFILE ${TGT${TGTID}_FILELIST})
```

Something like that maybe?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

**Requester Info**
Full name and company/organization if applicable

skliper, 2021-01-11:
@excaliburtb Do you need this in Caelum?

excaliburtb, 2021-01-11:
It would be really nice if it made it.

PR #1158 (2021-02-24): Fix #790, allow directories and symlinks in FILELIST.
**Describe the contribution**
Checks a name-based subdirectory under ${MISSION_DEFS} for files listed in the FILELIST for that target.  If file is a symlink, the link should be followed so the correct content is installed, not a symlink.

Fixes #790 

**Testing performed**
Created a symlink in a target name subdirectory (see additional context) and confirm that the correct file content is installed.

**Expected behavior changes**
Items in FILELIST can be in a target name directory, and may be symlinks.

**System(s) tested on**
Ubuntu 20.04
 [... cut]
commit 51a614c2 (2021-02-24): Fix #790, allow directories and symlinks in FILELIST.

Checks a name-based subdirectory under ${MISSION_DEFS} for
files listed in the FILELIST for that target.

If file is a symlink, the link should be followed so the correct
content is installed, not a symlink.
commit 9fd5d760 (2021-02-24): Merge pull request #1158 from jphickey/fix-790-install-symlink

Fix #790, allow directories and symlinks in FILELIST.
PR #1171 (2021-02-26): cFE Integration candidate: 2021-02-23
**Describe the contribution**
Fix #139, Do file writes in background
Fix #978, put task parameters into task record
Fix #790, allow directories and symlinks in FILELIST. 
Fix #735, add comment if null terminated or not.

Fix #1160, Shorten task info default filename 
Fix #1172, Update to use Ut_Stub_CheckDefaultReturnValue API  

**Testing performed**
See <https://github.com/nasa/cFS/pull/193/checks>

**Expected behavior changes**
PR #1148 - Implements a generic FS facility to perform file writes as a background job. Applications wanting to use this need t [... cut]
closed 2021-02-26 by astrogeco