nasa/cFE #802: network_includes.h needs to include sioLib.h for VxWorks target

mogar, 2020-08-12:
**Describe the bug**
When using RS-422 in io_lib under VxWorks, sioLib.h needs to be included.

**To Reproduce**
1. Use the 6.8.x tag to build a cFS system for VxWorks. 
2. Include io_lib in your apps
3. Ensure io_lib builds trans_rs422.c
4. Observe build failure:

```
[ 60%] Building C object apps/io_lib/CMakeFiles/io_lib.dir/fsw/src/services/trans_rs422.c.obj
<cfs>/apps/io_lib/fsw/src/services/trans_rs422.c: In function 'IO_TransRS422Init':
<cfs>/apps/io_lib/fsw/src/services/trans_rs422.c:111: error: 'SIO_HW_OPTS_SET' undeclared (first use in this function)
<cfs>/apps/io_lib/fsw/src/services/trans_rs422.c:111: error: (Each undeclared identifier is reported only once
<cfs>/apps/io_lib/fsw/src/services/trans_rs422.c:111: error: for each function it appears in.)
<cfs>/apps/io_lib/fsw/src/services/trans_rs422.c:111: error: 'CS8' undeclared (first use in this function)
<cfs>/apps/io_lib/fsw/src/services/trans_rs422.c:111: error: 'CLOCAL' undeclared (first use in this function)
<cfs>/apps/io_lib/fsw/src/services/trans_rs422.c:111: error: 'CREAD' undeclared (first use in this function)
make[7]: *** [apps/io_lib/CMakeFiles/io_lib.dir/fsw/src/services/trans_rs422.c.obj] Error 1
make[6]: *** [apps/io_lib/CMakeFiles/io_lib.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [CMakeFiles/cpu1_default_cpu1-all] Error 2
make[3]: *** [CMakeFiles/cpu1_default_cpu1-all.dir/all] Error 2
make[2]: *** [CMakeFiles/mission-all.dir/rule] Error 2
make[1]: *** [mission-all] Error 2
make: *** [all] Error 2
```

**Expected behavior**
Build should succeed.

**Code snips**
cfe/fsw/cfe-core/src/inc/network_includes.h should look like this (note the added sioLib.h line)

```
+++ b/fsw/cfe-core/src/inc/network_includes.h
@@ -42,6 +42,7 @@
     #include <arpLib.h>
     #include <netinet/in.h>
     #include <netinet/tcp.h>
+    #include <sioLib.h>
```

**System observed on:**
 - building under Ubuntu 18.04 host with VxWorks 6.9
 - Versions 
    - cFE 6.8.x
    - custom PSP and OSAL for PPC target
    - io_lib master

**Reporter Info**
Morgan Redfield, Astrobotic

jphickey, 2020-08-12:
This sounds like a bug in io_lib ... not really related to CFE ... unless I'm missing something?

Also worth noting the `sioLib.h` is not a standardized header file, it can't appear in any code expected to be cross-platform.

mogar, 2020-08-12:
Nope, it's in CFE. The io_lib app gets its network includes from a file that's in cfe, which is how it deal with cross-platform issues.

The file I noted in CFE (cfe/fsw/cfe-core/src/inc/network_includes.h) has some #ifdefs that specify platform and determine what headers get included. That's why sioLib.h needs to go here.

Take a look at that file in CFE and let me know if you have other concerns with this issue.

skliper, 2021-01-11:
@mogar - network_includes.h has been removed per #1067,  io_lib should include what it needs rather than depend on cFE for system specific includes (outside the scope of cFE).

closed 2021-03-30 by skliper

skliper, 2021-03-30:
Closed by #1067