nasa/cFE #803: Rename OMIT Deprecated flags with cFS codename instead of cFE version

astrogeco, 2020-08-14:
> does DEPRECATE_6_8 means that 6.8 was the last version which supports this function?

Yes, that is the paradigm we've been using up to this point -- the number reflects the last release were it was _not_ deprecated.  The idea of putting a version number in the tag was so we'd know which ones had "aged out" in a future release and should be taken out completely.  

However as I mentioned before now that we have a codename for API compatibility then maybe we should consider using that instead.

_Originally posted by @jphickey in https://github.com/nasa/cFE/pull/784#issuecomment-669204314_

skliper, 2020-08-14:
Might not be worth changing the old 6_7 over since that code can all be removed any time in this cycle.

jphickey, 2020-08-18:
See PR #817 -- removes all currently deprecated code blocks.

Probably don't really need to do anything - just come up with a codename-based convention for future deprecated blocks.

skliper, 2020-09-09:
Per the most recent versioning discussion - all the current deprecated code will be removed (or has been) as part of this major release development cycle.  Going forward the deprecate flags will not include a version number (like the osal model).

closed 2020-09-09 by skliper