nasa/cFE #808: Add cFE ES Information API functional tests

skliper, 2020-08-18:
**Is your feature request related to a problem? Please describe.**
Need open source functional tests for certifiability

**Describe the solution you'd like**
Add cFE Information API functional tests
– CFE_ES_GetResetType - Return the most recent Reset Type.
– CFE_ES_GetAppID - Get an Application ID for the calling Application.
– CFE_ES_GetAppIDByName - Get an Application ID associated with a specified Application name.
– CFE_ES_GetAppName - Get an Application name for a specified Application ID.
– CFE_ES_GetAppInfo - Get Application Information given a specified App ID.
– CFE_ES_GetTaskInfo - Get Task Information given a specified Task ID.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Requester Info**
Jacob Hageman - NASA/GSFC

wmoleski, 2020-08-19:
I am currently working on enhancing Joe Hickey's UT_ASSERT test to implement all of these API calls.

skliper, 2020-08-19:
Just for fun I was thinking maybe have test logic path where we could execute the first run as power on reset and tests should check everything except what is requires a processor reset... and could set up CDS or whatever... then at the end it could call a power on reset and check the final requirements.  On a system that autoboots this would be easier but on linux could execute from a script.  Just brainstorming.

PR #1251 (2021-04-08): Fix #808, ES API functional tests
**Describe the contribution**
Fixes #808 
New tests for the ES Info API's 

**Testing performed**
This is the test

**Expected behavior changes**
none

**Additional Context**
Dependant on #1268

Addresses are only valid on VxWorks system so the tests for when that is true have not be run and verified. 

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit 67ee18c4 (2021-04-08): Merge pull request #1251 from zanzaben/fix808_ES_API_Functional_Tests

Fix #808, ES API functional tests
commit e8a43be6 (2021-04-08): Fix #808, ES Info API Functional test
PR #1284 (2021-04-08): cFE Integration candidate: 2021-04-06
## Description

### PR #1251  

Fix #808, ES Info API Functional test 

Adds new tests for the ES Info APIs

## Context
Part of <https://github.com/nasa/cFS/pull/238>

## Tests
cFE Checks <https://github.com/nasa/cFE/pull/1284/checks>
Bundle Checks: <https://github.com/nasa/cFS/pull/238/checks>
closed 2021-04-08 by astrogeco