nasa/cFE #811: Add cFE Memory Manager API functional tests

skliper, 2020-08-18:
**Is your feature request related to a problem? Please describe.**
Need open source functional tests for certifiability

**Describe the solution you'd like**
Add cFE Memory Manager API functional tests
– CFE_ES_PoolCreate - Initializes a memory pool created by an application while using a semaphore during
processing.
– CFE_ES_PoolCreateEx - Initializes a memory pool created by an application with application specified block
sizes.
– CFE_ES_PoolCreateNoSem - Initializes a memory pool created by an application without using a
semaphore during processing.
– CFE_ES_GetPoolBuf - Gets a buffer from the memory pool created by CFE_ES_PoolCreate or CFE_ES -
_PoolCreateNoSem.
– CFE_ES_PutPoolBuf - Releases a buffer from the memory pool that was previously allocated via CFE_E -
S_GetPoolBuf.
– CFE_ES_GetMemPoolStats - Extracts the statistics maintained by the memory pool software.
– CFE_ES_GetPoolBufInfo - Gets info on a buffer previously allocated via CFE_ES_GetPoolBuf.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-08-18:
Question - should we consider finally implementing #56 as part of this?  Consolidating this code would simplify functional testing (one set or routines to implement, instead of two) as well as helping out with #651.  I had started implementing this a long time ago (CFE 6.5 timeframe) but never had the opportunity to finish/merge it, but it might still exist in one of my branches somewhere.

skliper, 2020-08-18:
Yes.  I'd think it would make sense to group all the mempool issues and resolve them.  A couple of them are required in the next release either way, so might as well refactor while we are at it to make everything simpler.

PR #1607 (2021-06-15): Fix #811, add Mempool functional tests
Describe the contribution
Fixes #811
adds new functional tests for es mempool

System(s) tested on
Ubuntu 20.04

Additional context
This is a just testing the nominal cases. More in depth tests will have to be added later.

Contributor Info - All information REQUIRED for consideration of pull request
Alex Campbell GSFC
commit 3ce33969 (2021-06-15): Fix #811, adding argument checks.
commit dc5018bf (2021-06-15): Fix #811, add Mempool functional tests
commit ff34a53b (2021-06-15): Merge pull request #1607 from zanzaben/fix811_Memory_Manager_Functional_Tests

Fix #811, add Mempool functional tests
closed 2021-06-16 by astrogeco